NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #4: ros2 topic cannot echo odometry topic

FPSychotic, 2021-11-14:
ros2 topic echo /visual_odometry/tf_stamped
Cannot echo topic '/visual_odometry/tf_stamped', as it contains more than one type: [isaac_ros_visual_odometry_interfaces/msg/VisualOdometryTransformStamped, nav_msgs/msg/Path]

closed 2021-11-14 by FPSychotic

willcmc, 2022-06-16:
@FPSychotic How did you sort this out?

FPSychotic, 2022-06-16:
@willcmc Really I don't remember what was the issue and even if it was in Xavier, I guess it was by the date.
Whatever it was, they are close to release a new version and will be a new version of Jetpack with it 4.5 , I would wait to the JP4.5 production release and try again.
I would investigate a little , but I have not a jetson nevermore.

willcmc, 2022-06-16:
I just solved the problem (with my setup) - apparently, RViz was looking for PoseStamped, and my node was publishing Pose. ros2 topic info was showing two message types too. Just made my node publish PoseStamped, working fine now.