NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #5: some basic questions

FPSychotic, 2021-11-14:
1-Hi, is it "odom" TF static against "base_link_somooth/rectified" TFs, like could be "map" TF compared with "camera_link" in realsenses?.

2-Can both TFs "base_link_somooth/rectified" be published simultaneosly working?
I mean will activate the rectified disconnect the smooth?

3-the odometry topic /visual_odometry/tf_stamped refer always to base_link_smooth TF?

4- recitified always appear in my my TF tree in Rviz2 even set in false, is this correct?

5- rectified include or means with loop closure?

thanks  by share it is a great peace of software, smart people

FPSychotic, 2021-11-15:
Thanks by your fast answer  and amazing job. 
Las question:
Am I right thinking that in case of both base_link are active the odometry topic refer to the rectified one? . 
Just show you how good your package works. In only VIO and applied to SLAM in RVIZ2 onboard as I didn't use the standard  installation as I didn't use docker. Low and very low light condition. Via VNC the slam video
https://youtu.be/V2Gz6kAzCzY
https://youtu.be/FWPwGUXdsIs

swapnesh-wani, 2021-11-18:
> Am I right thinking that in case of both base_link are active the odometry topic refers to the rectified one? 

`visual_odometry/tf_stamped` topic will always report both the `rectified_transform` and `smooth_transform`. In case of  `enable_rectified_pose = false`,  `rectified_transform` will be equal to `smooth_transform`. And, in case of  `enable_rectified_pose = true`, `rectified_transform` will report pose from loop closure and `smooth_transform` will report pose from pure visual odometry.

closed 2022-08-01 by hemalshahNV