NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #10: does it support rational_polynomial calibration from camera_info or config file?

FPSychotic, 2021-12-17:
i,m no able to make work a stereo camera, I think it is due to its rational_polynomial calibration , can be supported in a easy way? as config file?

hemalshahNV, 2021-12-17:
We do not yet support rational polynomial distortion model in Elbrus visual odometry. You can work around this by rectifying the image using isaac_ros_image_proc first and then passing in the undistorted image with pinhole model into Elbrus. Adding rational polynomial distortion model support is being prioritized for a future release.

FPSychotic, 2021-12-17:
Thanks, that workaround sounds nice to learn a little about, as I have no
experience with calibrations and so on. So time to learn a little.

Thanks!

FPSychotic, 2021-12-17:
Thanks a lot , I will try that workaround.
Thanks!

closed 2021-12-17 by FPSychotic

martinlucan, 2023-08-14:
@hemalshahNV Hi, has rational_polynomial calibration model been already released? Thank you.