NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #11: Can't Access RealSense D435i

Brac24, 2022-01-08:
I am running the Docker container from isaac_ros_common on Ubuntu 20.04 on my desktop not on a jetson. When running the launch file I get failed to set power state errors.

ros2 launch isaac_ros_visual_odometry isaac_ros_visual_odometry_realsense.launch.py
[INFO] [launch]: All log files can be found below /home/admin/.ros/log/2022-01-08-00-40-09-432131-pcmamba-2323
[INFO] [launch]: Default logging verbosity is set to INFO
/workspaces/isaac_ros-dev/install/isaac_ros_visual_odometry/share/isaac_ros_visual_odometry/launch/isaac_ros_visual_odometry_realsense.launch.py:16: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  realsense_camera_node = Node(
[INFO] [component_container-1]: process started with pid [2336]
[INFO] [realsense2_camera_node-2]: process started with pid [2338]
[realsense2_camera_node-2] [INFO] [1641602409.565854679] [RealSenseCameraNode]: RealSense ROS v3.2.3
[realsense2_camera_node-2] [INFO] [1641602409.565879440] [RealSenseCameraNode]: Built with LibRealSense v2.50.0
[realsense2_camera_node-2] [INFO] [1641602409.565883781] [RealSenseCameraNode]: Running with LibRealSense v2.50.0
[realsense2_camera_node-2] [WARN] [1641602409.575876964] [RealSenseCameraNode]: Device 1/1 failed with exception: failed to set power state
[realsense2_camera_node-2] [ERROR] [1641602409.575892514] [RealSenseCameraNode]: The requested device with  is NOT found. Will Try again.
[component_container-1] [INFO] [1641602409.757698050] [visual_odometry_launch_container]: Load Library: /workspaces/isaac_ros-dev/install/isaac_ros_visual_odometry/lib/libvisual_odometry_node.so
[component_container-1] [INFO] [1641602409.789745607] [visual_odometry_launch_container]: Found class: rclcpp_components::NodeFactoryTemplate<isaac_ros::visual_odometry::VisualOdometryNode>
[component_container-1] [INFO] [1641602409.789775598] [visual_odometry_launch_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<isaac_ros::visual_odometry::VisualOdometryNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/visual_odometry_node' in container '/visual_odometry_launch_container'
[realsense2_camera_node-2] [WARN] [1641602415.585547627] [RealSenseCameraNode]: Device 1/1 failed with exception: failed to set power state
[realsense2_camera_node-2] [ERROR] [1641602415.585565458] [RealSenseCameraNode]: The requested device with  is NOT found. Will Try again.

hemalshahNV, 2022-01-08:
Did you happen to plug your Realsense in after you started the dev container? If so, the camera device may not have existed and so did not get mapped in to the container while starting. Restarting your container with the camera plugged in would fix this if that is the problem.

Brac24, 2022-01-08:
Hello @hemalshahNV thank you for the reply.I don't think I did but as a sanity check just restarted everything leaving the camera plugged in but it still gives me the same error. I found this comment https://github.com/IntelRealSense/realsense-ros/issues/1408#issuecomment-768007415 but not sure if this would break other things the container needs.

And what's weird is this is the error I get when running rs-depth in the container:
There are 1 connected RealSense devices.
rs_error was raised when calling rs2_create_device(info_list:0x559f299a51f0, index:0):
    failed to set power state

Somehow it does see the device but can't turn it on.

hemalshahNV, 2022-01-08:
Just a guess, but perhaps the camera is underpowered because it is connected to an unpowered port?

Brac24, 2022-01-08:
Yeah I did think about that a bit. I did try a different port but it gave me the same thing. But I'll try all the usb ports I have on here to verify.

Brac24, 2022-01-08:
So I verified all ports and was still getting the same issue. I can power it on just fine on my windows laptop. I upgraded the firmware of the camera through my laptop and tried again but didn't seem to make a difference. Could this be an access issue trying to connect to the camera through docker? But it shows that it detects the camera but fails to power it apparently so I assume it's not an access issue.

Brac24, 2022-01-09:
Ok so I've solved it by installing the librealsense sdk on my machine. Not sure if that was a prerequisite for this to work so I apologize if I missed that.

closed 2022-01-09 by Brac24