NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #18: Could not find a package configuration file provided by "vpi"

MadlyFX, 2022-03-27:
Using Ubuntu 20.04.4, and ROS2 Foxy. Following the steps in the readme, but build fails with this error:

```
colcon build --symlink-install
Starting >>> isaac_ros_common
Starting >>> isaac_ros_test
Starting >>> isaac_ros_nvengine_interfaces
Starting >>> isaac_ros_visual_slam_interfaces
Finished <<< isaac_ros_test [0.93s]                                  
--- stderr: isaac_ros_common                                         
CMake Error at CMakeLists.txt:31 (find_package):
  By not providing "Findvpi.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "vpi", but
  CMake did not find one.

  Could not find a package configuration file provided by "vpi" with any of
  the following names:

    vpiConfig.cmake
    vpi-config.cmake

  Add the installation prefix of "vpi" to CMAKE_PREFIX_PATH or set "vpi_DIR"
  to a directory containing one of the above files.  If "vpi" provides a
  separate development package or SDK, be sure it has been installed.

```

hemalshahNV, 2022-03-27:
Are you using the Docker container through run_dev.sh in isaac_ros_common or running natively? You’ll need to install the VPI debians provided in isaac_ros_common for x86 of running outside of Docker.

MadlyFX, 2022-03-27:
Running natively, not Docker. I'll have a look at the other library. 

Should I be sourcing opt/ros/foxy/setup.bash before building as well?

hemalshahNV, 2022-03-27:
Take a look at the instructions in isaac_ros_common for installing VPI for x86 to work natively or consider using the Docker container where we were able to configure more the environment for you.

Yes, you would always need to source the ROS distro before building as usual.

engcang, 2022-04-12:
> Are you using the Docker container through run_dev.sh in isaac_ros_common or running natively? You’ll need to install the VPI debians provided in isaac_ros_common for x86 of running outside of Docker.

Hi. @hemalshahNV 
I am also trying to run this repo outside the docker container.
Where can I find that specific VPI debian in `isaac_ros_common`? Can you give me the link or path of it? I cannot find it.
I installed VPI version 2.0.9 using SDK manager and couldn't make this repo built. 
Is this repo only compatible with VIP 1.1.11?

hemalshahNV, 2022-04-13:
The Debians for VPI 1.1.11 x86 can be found in isaac_ros_common/docker/vpi. Isaac ROS packages support Jetpack 4.6.1(aarch64)/Ubuntu 20.04(x86) with VPI 1.1.11 currently. An upcoming release targeting June will bring JP 5(aarch64) support and updated to use the new APIs in VPI 2.0.9.

engcang, 2022-04-14:
@hemalshahNV I found it there. Thanks.
What are the corresponding files in this link? https://repo.download.nvidia.com/jetson/
maybe [libnvvpi1_1.1.11_amd64.deb]
https://repo.download.nvidia.com/jetson/x86_64/bionic/pool/main/libn/libnvvpi1/libnvvpi1_1.1.11_amd64.deb) and [vpi1-dev_1.1.11_amd64.deb](https://repo.download.nvidia.com/jetson/x86_64/bionic/pool/main/v/vpi1-dev/vpi1-dev_1.1.11_amd64.deb) both? even these files are for X86_64/Bionic (not Focal Fossa)

closed 2022-08-01 by hemalshahNV