NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #20: IMU integration

MartinSedlacek, 2022-04-07:
Hello,
I'm trying to test isaac_ros_visual_slam with IMU (the one inside realsense d455 module) using realsense launch example. 

**I edited  [realsense example](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/main/isaac_ros_visual_slam/launch/isaac_ros_visual_slam_realsense.launch.py) like this:** 

<details>
  <summary>Edited launch</summary>
  
  ```   
  visual_slam_node = ComposableNode(
        name='visual_slam_node',
        package='isaac_ros_visual_slam',
        plugin='isaac_ros::visual_slam::VisualSlamNode',
        parameters=[{
                    'enable_rectified_pose': True,
                    'denoise_input_images': False,
                    'rectified_images': True,
                    'enable_debug_mode': False,
                    'debug_dump_path': '/tmp/elbrus',
                    'enable_slam_visualization': True,
                    'enable_landmarks_view': True,
                    'enable_observations_view': True,
                    'map_frame': 'map',
                    'odom_frame': 'odom',
                    'base_frame': 'camera_link',
                    'enable_imu': True,
                    'input_left_camera_frame': 'camera_infra1_frame',
                    'input_right_camera_frame': 'camera_infra2_frame',
                    'input_imu_frame': 'camera_accel_frame',
                    'gravitational_force': [-0.22555294, -9.83607, 0.049]
                    }],
        remappings=[('stereo_camera/left/image', 'infra1/image_rect_raw'),
                    ('stereo_camera/left/camera_info', 'infra1/camera_info'),
                    ('stereo_camera/right/image', 'infra2/image_rect_raw'),
                    ('stereo_camera/right/camera_info', 'infra2/camera_info'),
                    ('visual_slam/imu', '/imu')]
    )
  ```
</details>

- 'gravitational_force': [-0.22555294, -9.83607, 0.049] -> gravitational_force was changed from default value as stated [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam#available-components-for-visual_slam_node)  [-0.22555294, -9.83607, 0.049] is reading from IMU at the start of trajectory estimation

- 'input_imu_frame': 'camera_accel_frame' -> input_imu_frame was set to available one from tf tree [following this](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam#coordinate-frames) 
<details>
  <summary>Visualized here</summary>
  
[image]

</details>

**My issue:** 
Although integrator_states changes to 3 (at launch it as 2 and later 3) I cannot see any improvements in estimated trajectory. (IMU measurements seems to be relatively in order)

In a experiment, where I simulate lost of image (the device is without movement) estimated trajectory does drift (not small drift caused by noise integration but a big one).

**Questions:** 
I suspect that my input_imu_frame or gravitational_force parameters are not in order, could you provide some more insight how to set these parameters?

What should I expect from IMU integration? Overall better trajectory estimation? Or IMU is used only in situations where image is invalid?

Is there a simple way how to test whether IMU integration is in order?

swapnesh-wani-nvidia, 2022-08-01:
In order to verify the integration of IMU, you could try to "blind" the camera imagers and try small rotational changes about principal axes and see if the poses are still coming and are expected. 

For the gravitational vector, it looks like it is aligned to the camera coordinate frame so the Z component of the gravitational vector should have the value 9.81

closed 2022-09-19 by hemalshahNV

chivas1000, 2022-11-24:
> Hello, I'm trying to test isaac_ros_visual_slam with IMU (the one inside realsense d455 module) using realsense launch example.
> 
> **I edited [realsense example](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/main/isaac_ros_visual_slam/launch/isaac_ros_visual_slam_realsense.launch.py) like this:**
> 
> Edited launch
> * 'gravitational_force': [-0.22555294, -9.83607, 0.049] -> gravitational_force was changed from default value as stated [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam#available-components-for-visual_slam_node)  [-0.22555294, -9.83607, 0.049] is reading from IMU at the start of trajectory estimation
> * 'input_imu_frame': 'camera_accel_frame' -> input_imu_frame was set to available one from tf tree [following this](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam#coordinate-frames)
> 
> Visualized here
> **My issue:** Although integrator_states changes to 3 (at launch it as 2 and later 3) I cannot see any improvements in estimated trajectory. (IMU measurements seems to be relatively in order)
> 
> In a experiment, where I simulate lost of image (the device is without movement) estimated trajectory does drift (not small drift caused by noise integration but a big one).
> 
> **Questions:** I suspect that my input_imu_frame or gravitational_force parameters are not in order, could you provide some more insight how to set these parameters?
> 
> What should I expect from IMU integration? Overall better trajectory estimation? Or IMU is used only in situations where image is invalid?
> 
> Is there a simple way how to test whether IMU integration is in order?

Hi, did you succefully set imu fusion in visual slam?
I'm also using d435i to do this, but I do not know which frame is "inpu_imu_frame", since the expected imu frame of d435i is camera_imu_optical_frame, but it doesn't show in TF trees although it does output /camera/imu topics...
any assistance or advice would be appreciated, thanks

mzahana, 2022-12-26:
Hi @MartinSedlacek ! 
I am also trying to use the D455's IMU , but I am not sure about the correct parameters values. Have you figured this out?