NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #22: Bag file can not be used

engcang, 2022-04-14:
Hi.

With recorded bag files including camera info topics and stereo images, the SLAM does not work.

It generates following error
`[ERROR]: Transform is impossible. canTransform(camera_infra1_frame->camera_link) returns false` 

With live topics, the SLAM works super well. 

Thanks.

hemalshahNV, 2022-04-14:
Is the tf topic being recorded in your bag? The static transform are likely not available if I read the error correctly for the frames on the camera (the imager and the camera frame).

engcang, 2022-04-14:
@hemalshahNV 
Thanks for your reply.
I thought the SLAM node only subscribes stereo images and camera info topics by looking into `rqt_graph`.

Let me try to record `/tf_static` as well. I will be back soon.

engcang, 2022-04-16:
For anyone who wants to try this algorithm with recorded `bag` files,
since recording and replaying `/tf_static` is not implemented in `ROS2`,
Use `tf2_ros` package's `static_transform_publisher` node as [my launch file](https://github.com/engcang/vins-application/blob/master/NVIDIA_Isaac_Elbrus/elbrus.launch.py).

You can see the result video here: https://youtu.be/V24_xtkscjc

closed 2022-08-01 by hemalshahNV