NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #24: it won't build on Jetson JP4.6.1 but for isaac common script

AndreV84, 2022-04-21:
```
-- stderr: isaac_ros_nvengine                                                                                                
/usr/bin/ld:/workspaces/isaac_ros-dev/isaac_ros_nvengine/gxf/lib/gxf_jetpack46_1/core/libgxf_core.so: file format not recognized; treating as linker script
/usr/bin/ld:/workspaces/isaac_ros-dev/isaac_ros_nvengine/gxf/lib/gxf_jetpack46_1/core/libgxf_core.so:1: syntax error
collect2: error: ld returned 1 exit status
make[2]: *** [libgxe_node.so] Error 1
make[1]: *** [CMakeFiles/gxe_node.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< isaac_ros_nvengine [22.3s, exited with code 2]
```
after performing the isaac common image script I added src then cloned the repository; then executed colcon build
it won't work

hemalshahNV, 2022-04-21:
Make sure to have git-lfs installed and run git lfs pull​ to download the binaries into place if you did not have it installed before cloning.

AndreV84, 2022-04-21:
@hemalshahNV 
Thank you for the update.
after git lfs pull it seems got build 
```
Finished <<< isaac_ros_test [5.03s]
Starting >>> isaac_ros_nvengine
Starting >>> isaac_ros_visual_slam               
Finished <<< isaac_ros_nvengine [2.00s]                                                                          
[Processing: isaac_ros_visual_slam]                             
[Processing: isaac_ros_visual_slam]                                     
[Processing: isaac_ros_visual_slam]                                       
[Processing: isaac_ros_visual_slam]                                       
Finished <<< isaac_ros_visual_slam [2min 3s]                             

Summary: 6 packages finished [2min 9s]
  1 package had stderr output: isaac_ros_test

```
However, how do I run it for zed mini camera? I got to execute every time to run the container scripts/run_dev.sh from ros common isaac?

AndreV84, 2022-04-22:
@hemalshahNV 
Could you publish the zed launch file, please?
Previously ther ewas https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_odometry/blob/main/isaac_ros_visual_odometry/launch/isaac_ros_visual_odometry_zed2.launch.py
which is not present.
Moreover, could you share launch file for zed [  not zed2 but just zed]
Thanks
AV

AndreV84, 2022-04-22:
@hemalshahNV 
lfs pull helped building container from script but therre still seems something missed?
```
nvidia@nvidia-desktop:~/isaac_ros_common/src/isaac_ros_visual_slam$ git lfs pull
nvidia@nvidia-desktop:~/isaac_ros_common/src/isaac_ros_visual_slam$ cd ..
nvidia@nvidia-desktop:~/isaac_ros_common/src$ cd ..
nvidia@nvidia-desktop:~/isaac_ros_common$ ./scripts/run_dev.sh 
isaac_ros_dev not specified, assuming /home/nvidia/isaac_ros_common
~/isaac_ros_common ~/isaac_ros_common
Attaching to running container: isaac_ros_dev-aarch64-container
admin@nvidia-desktop:/workspaces/isaac_ros-dev$ . /opt/ros/foxy/setup.bash 
admin@nvidia-desktop:/workspaces/isaac_ros-dev$ . install/setup.bash 
admin@nvidia-desktop:/workspaces/isaac_ros-dev$ ros2 launch isaac_ros_visual_odometry s isaac_ros_visual_odometry_zed2.launch.py
scripts/ src/     
admin@nvidia-desktop:/workspaces/isaac_ros-dev$ ros2 launch isaac_ros_visual_odometry s isaac_ros_visual_odometry_zed2.launch.py
scripts/ src/     
admin@nvidia-desktop:/workspaces/isaac_ros-dev$ ros2 launch isaac_ros_visual_odometry s isaac_ros_visual_odometry_zed2.launch.py
scripts/ src/     
admin@nvidia-desktop:/workspaces/isaac_ros-dev$ ros2 launch isaac_ros_visual_odometry isaac_ros_visual_odometry_zed2.launch.py
Package 'isaac_ros_visual_odometry' not found: "package 'isaac_ros_visual_odometry' not found, searching: ['/workspaces/isaac_ros-dev/install/isaac_ros_visual_slam', '/workspaces/isaac_ros-dev/install/isaac_ros_visual_slam_interfaces', '/workspaces/isaac_ros-dev/install/isaac_ros_test', '/workspaces/isaac_ros-dev/install/isaac_ros_nvengine', '/workspaces/isaac_ros-dev/install/isaac_ros_nvengine_interfaces', '/workspaces/isaac_ros-dev/install/isaac_ros_common', '/opt/ros/foxy/install']"
admin@nvidia-desktop:/workspaces/isaac_ros-dev$ ros2 launch isaac_ros_visual_odometry isaac_ros_visual_odometry_zed2.launch.py

hemalshahNV, 2022-06-06:
The package has been renamed `isaac_ros_visual_slam`.

closed 2022-08-01 by hemalshahNV