NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #25: Working with Isaac Sim does not work well

SnowMasaya, 2022-04-30:
I walked through the following link instruction

- https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam#working-with-isaac-sim

But it does not work well

```
ros2 launch isaac_ros_visual_slam isaac_ros_visual_slam_isaac_sim.launch.py
```

The following error happened. And I did not confirm the ros message.

```
[INFO] [launch]: All log files can be found below /home/admin/.ros/log/2022-04-30-04-43-54-457934-omn-a40-ubuntu-2718564
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [2718584]
[component_container-1] [INFO] [1651293834.889712235] [visual_slam_launch_container]: Load Library: /workspaces/isaac_ros-dev/ros_ws2/install/isaac_ros_visual_slam/lib/libvisual_slam_node.so
[ERROR] [component_container-1]: process has died [pid 2718584, exit code -4, cmd '/opt/ros/foxy/lib/rclcpp_components/component_container --ros-args -r __node:=visual_slam_launch_container -r __ns:=/'].
```

Could you tell me how to solve this problem?

hemalshahNV, 2022-04-30:
Which platform? Are you running within the isaac_ros_common container or natively? Do you know which version of the CUDA toolkit you have installed? Any information about how you’re running it may help clarify the problem.

  *   https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam#working-with-isaac-sim

SnowMasaya, 2022-05-03:
The information is followings

- isaac_ros_common container 
- OS: Ubuntu 20.04
- NVIDIA driver: 470.82.01
- cuda: 11.4

hemalshahNV, 2022-08-01:
Could you please try again with the latest Isaac ROS Visual SLAM and Isaac ROS Common?

romi2002, 2022-09-07:
Hey, I'm having the same issue, with the latest ROS Visual SLAM and Issac ROS Common releases. Are there any steps I could take to debug this issue?
```
[component_container-1] [INFO] [1662589262.231587091] [visual_slam_launch_container]: Load Library: /workspaces/isaac_ros-dev/install/isaac_ros_visual_slam/lib/libvisual_slam_node.so
[ERROR] [component_container-1]: process has died [pid 26150, exit code -4, cmd '/opt/ros/humble/install/lib/rclcpp_components/component_container --ros-args -r __node:=visual_slam_launch_container -r __ns:=/'].```
with:
* issac_ros_common container
* OS: Ubuntu 20.04
* Cuda 11.7
* Nvidia driver: 515.65.01

muratkoc503, 2022-09-09:
@romi2002 @hemalshahNV I have similar issue. [component-container: process has died.](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/issues/47)
Also Galactic version get this error too. My computer(not docker container) doesnt include /opt/ros/humble/install/ directory. 

[image]

hemalshahNV, 2022-09-12:
@romi2002 Could you tell us what GPU you're running (the output of `nvidia-smi` would be helpful)? Which versions of Isaac Sim are you running? Isaac Sim 2022.1.1 and Isaac ROS DP1.1 v0.11.0 should work well together.

romi2002, 2022-09-21:
`nvidia-smi` returns the following, 
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 515.65.01    Driver Version: 515.65.01    CUDA Version: 11.7     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  On   | 00000000:84:00.0  On |                  N/A |
| 10%   53C    P2    46W / 215W |   1553MiB /  8192MiB |     12%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
```

Where can I check the Issac Sim version? I'm following the steps in the readme, running the sim inside docker with:

```
cd ~/workspaces/isaac_ros-dev/src/isaac_ros_common && \
  ./scripts/run_dev.sh```

hemalshahNV, 2022-09-26:
You'll want to run Isaac Sim on the host (not inside the Docker container).

closed 2022-10-24 by jaiveersinghNV