NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #35: wget: unable to resolve host address 'developer.download.nvidia.com'

hpf9017, 2022-07-03:
--2022-07-03 15:27:56--  https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/cuda-keyring_1.0-1_all.deb
Resolving developer.download.nvidia.com (developer.download.nvidia.com)... failed: Temporary failure in name resolution.
wget: unable to resolve host address 'developer.download.nvidia.com'
The command '/bin/bash -c apt-key del 7fa2af80 && mkdir -p /tmp && cd /tmp         && wget https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/cuda-keyring_1.0-1_all.deb         && dpkg -i cuda-keyring_1.0-1_all.deb         && rm cuda-keyring_1.0-1_all.deb         && add-apt-repository --remove 'deb https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64/ /'' returned a non-zero code: 4
Failed to build base image: isaac_ros_dev-x86_64, aborting.
~/your_ws/src/isaac_ros_common

swapnesh-wani-nvidia, 2022-07-06:
Hi @hpf9017, could you please provide more details on the steps to reproduce this issue? Also, please share your system configuration, environment setup (are you using docker or it is a bare-metal setup), and network configuration (firewall configuration).

hpf9017, 2022-07-07:
@swapnesh-wani-nvidia this is maybe the network problem. I have run the demo with rosbag replay following the instructions of the humble and Jetpack 5.0.1 version. Thank you very much !

closed 2022-07-07 by hpf9017