NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #36: Container build fail

demul, 2022-07-06:
I follow the quickstart Instruction
and
when i launch the docker container using the run_dev.sh script,
it fail with message below
 
```c
Starting >>> interactive_markers
Starting >>> nav2_util
Starting >>> image_rotate
Finished <<< image_proc [1min 17s]
Starting >>> stereo_image_proc
Finished <<< image_view [1min 19s]
Finished <<< tf2_eigen [1min 1s]
Starting >>> depth_image_proc
Starting >>> geometry2
Finished <<< geometry2 [14.1s]
Finished <<< image_rotate [38.9s]
--- stderr: interactive_markers
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/test_interactive_marker_server.dir/build.make:76: CMakeFiles/test_interactive_marker_server.dir/test/interactive_markers/test_interactive_marker_server.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:242: CMakeFiles/test_interactive_marker_server.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< interactive_markers [1min 58s, exited with code 2]
[Processing: depth_image_proc, interactive_markers, nav2_util, robot_state_publisher, rosbag2_transport, rviz_common, stereo_image_proc]
[Processing: depth_image_proc, interactive_markers, nav2_util, robot_state_publisher, rosbag2_transport, rviz_common, stereo_image_proc]
Aborted  <<< stereo_image_proc [2min 1s]
Aborted  <<< robot_state_publisher [4min 5s]
Aborted  <<< rosbag2_transport [4min 27s]
Aborted  <<< nav2_util [4min 7s]
Aborted  <<< depth_image_proc [3min 57s]
Aborted  <<< rviz_common [4min 42s]
```

I'm using Ubuntu 18.04 WSL.
Help me.

hemalshahNV, 2022-07-06:
We have never tried to build the image on WSL, but ROS2 Humble won’t build on Ubuntu 18.04 (which is what is happening when the crash here occurs). You may want to try native Ubuntu 20.04 instead.

demul, 2022-07-06:
I can't understand.
Maybe I have not tried to build ROS2 Humble on my native WSL Ubuntu 18.04.
ROS2 Humble would be built in the provided docker image(launched by run_dev.sh).
and maybe the provided docker image is Ubuntu 20.04 right?

Is there any misunderstanding i have about your quickstart build system?

demul, 2022-07-14:
@hemalshahNV @hemalshahNV @hemalshahNV @hemalshahNV @hemalshahNV @hemalshahNV @hemalshahNV @hemalshahNV @hemalshahNV @hemalshahNV @hemalshahNV @hemalshahNV @hemalshahNV 
I can't understand.
Maybe I have not tried to build ROS2 Humble on my native WSL Ubuntu 18.04.
ROS2 Humble would be built inside the provided docker container(launched by run_dev.sh).
and maybe the provided docker image is Ubuntu 20.04. Right?

It sounds strange that There is os-dependency, although this package is distributed by docker.

Is there any points i misunderstand about your quickstart build system?

hemalshahNV, 2022-07-14:
WSL is not supported for building the container. Please consider building on a native Ubuntu system.

closed 2022-07-15 by hemalshahNV