NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #43: "Tracker is lost" on KITTI rosbag

hemalshahNV, 2022-08-01:
Alright, i was able to launch example code using docker. It works. But I'm still getting nowhere trying to launch one of the kitti rosbags using `ros2 bag play ../datasets/kitti/rosbags/ros2/kitti_2011_09_29_drive_0071_synced/ --remap /kitti/camera_gray_right/image_raw:=/stereo_camera/right/image /kitti/camera_gray_left/image_raw:=/stereo_camera/left/image /kitti/camera_gray_left/camera_info:=/camera_info_left /kitti/camera_gray_right/camera_info:=/camera_info_right` command. Still "Tracker is lost" issue.

_Originally posted by @DaddyWesker in https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/issues/37#issuecomment-1188968363_

hemalshahNV, 2022-08-01:
@DaddyWesker Could you tell us where you obtained this KITTI rosbag from? There may be missing frames in the recording that can lead to "tracker lost" issues. [Here](http://www.cvlibs.net/datasets/kitti/eval_odometry_detail.php?&result=252948fc1b95f8b635cf63df526e4864cb6c0f41) are the official results for Isaac Visual SLAM on KITTI.

DaddyWesker, 2022-08-02:
Well, I've downloaded a sequence from kitti website, used kitti2bag to make a rosbag (ros1 bag) which I've used with other SLAM systems and it worked fine. 
Then, since your isaac bis based on ROS2 i needed to convert ros1 bag to ros2 bag. I've used these two repos for that purpose. 

https://github.com/leo-drive/ros1_bridge
https://github.com/leo-drive/proto_to_rosbag2

swapnesh-wani-nvidia, 2022-08-08:
Are you seeing "Tracker is lost" frequently or just at the beginning? Also, could you verify if the output of your rosbag and output poses from Isaac ROS Visual SLAM are correlated? Do you see any sudden jumps in the image sequence (from your rosbag)? This is just to verify if there are any skipped frames or timestamp issues that could lead to a "Tracker is lost" scenario.

Corufa, 2022-12-04:
I prepared the KITTI dataset in the same way as @DaddyWesker  and ran the sample code. However, I frequently get the message "Tracker is lost". rviz confirmed that there are no sudden jumps in the image sequence.