NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #44: Elbrus do not publishes the topic /map. How can i use it with nav2?

RailWar, 2022-08-05:
nav2 uses openvlsam is default. But I want use this project. But I get the message /map topic is not published. Can i use Elbrus and nav2? Or is there other engines for navigation? Are there examples for this? My ros2 version is humble.

hemalshahNV, 2022-08-05:
We integrate Isaac VSLAM into Nav2 as an odometry source but not as a map/global localization source yet. What does your sensor setup look like and what workflow you're trying to build? The Nav2 community was developing full visual mapping & navigation using OpenVSLAM if I recall but I'm not sure what the status is. OpenVSLAM had shut down in favor of OrbSLAM at some point.

RailWar, 2022-08-09:
I have to connect up to 4 Zed 2i cameras (instead high-cost lidar)  to Jetson Xavier NX and get the navigation. Because Elbrus don't uses CUDA in Isaac SDK I want to make the pipe: ros2 = Zed 2i + Elbrus + nav2

hemalshahNV, 2022-08-09:
In our upcoming release for Isaac ROS, we should have an example in Nvblox to use with Isaac VSLAM . Along with Nav2, you should have all of the building blocks for a visual navigation solution. We're still evaluating this on our reference robots, however, and your experience would be greatly appreciated.

hemalshahNV, 2022-09-19:
Please take a look at the Nvblox/Isaac VSLAM tutorial here: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/blob/main/docs/tutorial-nvblox-vslam-realsense.md.

closed 2022-09-19 by hemalshahNV