NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #47: isaac_ros_visual_slam as simulation with Gazebo return [CUDA] failure

muratkoc503, 2022-09-05:
Hi,
I am working with isaac_ros_visual_slam package and Gazebo. 
[image]

My error:
[component_container-1] [INFO] [1662379434.366503061] [visual_slam_node]: Use use_gpu: true
[component_container-1] [ERROR] 140409917632320 [CUDA] failure: 13, file: /home/akorovko/Code/elbrus/src/modules/cuda_modules/src/lk_tracker.cpp, line: 9

Also, i use pytorch environment with cuda.

Thanks for reply.

muratkoc503, 2022-09-06:
I work on Ros Galactic. Maybe this is reason of my error. i dont know.

hemalshahNV, 2022-09-06:
Could you describe the platform you're running on? Are you running within the Isaac ROS Dev base Docker container or natively on host? It looks like an x86 machine running ROS2 Humble (not Galactic which is incompatible) but with which NVIDIA GPU available for CUDA?

muratkoc503, 2022-09-06:
My knowledge is not good at docker. So, sorry. I work on 64 bit on computer.(laptop). CUDA Version: 11.7. I will look tomorrow. Thanks for Humble case.
**nvidia-smi:**
[image]

muratkoc503, 2022-09-08:
I try to run rosbag example which is "small_pool_test". But, it give same error. My GPU is NVIDIA GeForce MX110. Is it enough for isaac_ros_visual_slam package? Also, i try path as "nvidia-container-runtime" and "/usr/build/nvidia-container-runtime", but this is not work. I install nvidia-container-runtime package is seen belove: I use Ubuntu focal(20.04) and jammy(22.04). But bionic version is installed. I dont know, what is the reason of my error. I doubt from my nvidia-container-runtime. Also i try Ros2 Galactic and Humble.

[image]

[image]

[image]

muratkoc503, 2022-09-09:
Also, i install nvidia-container-runtime to native computer. I dont install nvidia-container-runtime to this(isaac_ros_visual_slam) Docker container. Does the nvidia-container-runtime installing in the Docker container is necessary?

hemalshahNV, 2022-09-12:
It appears you're running this on a Maxwell architecture GPU. We think this may not be recent enough for Isaac ROS Visual SLAM to work with. Can you try running on a Volta or newer GPU? We'll work on a more graceful failure message if this is indeed the problem you're running into.

muratkoc503, 2022-09-13:
I am not working with Gazebo, i am running bag file now:

`source /workspaces/isaac_ros-dev/install/setup.bash && \
   ros2 bag play src/isaac_ros_visual_slam/isaac_ros_visual_slam/test/test_cases/rosbags/small_pol_test/`

Also, My Gazebo doesn't use GPU. Probably i dont use GPU correctly.

**watch nvidia-smi:** (on natural computer)
[image]

**watch nvidia-smi:** (in docker container)
[image]

**top:** (in docker container)
[image]

component container, rviz2 and all other program doesn use GPU. Only X-server and office program use GPU.

[image]

Also, i have 2GB memory. But only 3MB is used. Gazebo doesn't use GPU. Accually, i have a nvidia-driver
I try to solve GPU using with program (ex: Gazeob).

muratkoc503, 2022-09-14:
I think, my installation is correct. Because i test my GPU(Nvidia Container) at my native computer and any docker container.

**Run GPU in native computer:** (BlackScholes sample)
[image]

**Run GPU in docker container:** (ros_caffe sample)
[image]

Also, i run optional command after installing isaac_ros_visual_slam package:
`colcon test --executor sequential
`

This command get below error:
```
Starting >>> isaac_ros_visual_slam
--- stderr: isaac_ros_visual_slam                         
Errors while running CTest
Output from these tests are in: /workspaces/isaac_ros-dev/build/isaac_ros_visual_slam/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
Finished <<< isaac_ros_visual_slam [18.2s]	[ with test failures ]
```

So, i read log file. i see 3 errror message in : colcon test --executor sequential command

```
[component_container-1] [ERROR] 140288154709824 [CUDA] failure: 13, file: /home/akorovko/Code/elbrus/src/modules/cuda_modules/src/lk_tracker.cpp, line: 9
[ERROR] [component_container-1]: process has died [pid 44573, exit code -6, cmd '/opt/ros/humble/install/lib/rclcpp_components/component_container --ros-args -r __node:=visual_slam_container -r __ns:=/'].
[ERROR] 140288154709824 [CUDA] failure: 13, file: /home/akorovko/Code/elbrus/src/modules/cuda_modules/src/lk_tracker.cpp, line: 9
```
Also, i didn't can't try with Volta or new GPU. I have only this computer. Probably, my computer doesn't support this. I don't know. I have 2GB GPU memory.

muratkoc503, 2022-09-14:
I try to Ros BAG sample in **isaac_ros_nvblox** package (nvblox_pol sample).
It work correct for me.

**Result:**
**Run GPU isaac_ros_nvblox:**
[image]

**Rviz Result:**
[image]

Probably, [CUDA] failure: 13, file: /home/akorovko/Code/elbrus/src/modules/cuda_modules/src/lk_tracker.cpp, line: 9 error interested in isaac_ros_visual_slam package. isaac_ros_nvblox work my computer with GPU.

jaiveersinghNV, 2022-10-03:
Isaac ROS Visual SLAM is only compatible with Volta and newer GPU architectures.

closed 2022-10-10 by swapnesh-wani-nvidia

madgrizzle, 2023-10-21:
Can someone explain why a Volta (or newer) GPU is required?  I've installed and have successfully run this on a Tesla P4 that is based on Pascal.  I've yet to incorporate it into my robot, but what specifically does the Volta architecture provide that's missing on the Pascal that's in use by visual slam?