NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #51: Repository not working

andresancal, 2022-10-18:
The ament_cmake can not be found.
I have already tried to source the opt/ros/humble/setup.bash 

[image]

hemalshahNV, 2022-10-18:
Are you using the Isaac ROS Dev base images or a custom install of ROS2 Humble? Can you build any package if colcon can't find ament_cmake?

jaiveersinghNV, 2022-10-24:
Could you please run: `ros2 pkg prefix ament_cmake_auto`, and see if ROS2 is able to find this path installed? When I run this inside the Docker container on my machine, I get the expected installation path of `/opt/ros/humble/install`.

jn128c, 2022-11-02:
Had the same issue, fix your folder/file permissions. Or try sudo.

closed 2022-11-22 by hemalshahNV