NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #53: colcon build error

chivas1000, 2022-10-21:
Hi I'm using the VSLAM,
but got this error when colcon building:

admin@ubuntu-Default-string:/workspaces/isaac_ros-dev$ colcon build --symlink-install
Starting >>> nvblox_msgs
Starting >>> realsense2_camera_msgs
Starting >>> nvblox
Starting >>> isaac_ros_test
Starting >>> isaac_ros_visual_slam_interfaces
Starting >>> nvblox_cpu_gpu_tools                                                                                                  
Starting >>> nvblox_performance_measurement_msgs
Starting >>> isaac_ros_apriltag_interfaces
Starting >>> isaac_ros_bi3d_interfaces
Starting >>> isaac_ros_common
Starting >>> isaac_ros_tensor_list_interfaces                                                                                       
Starting >>> nvblox_isaac_sim
Finished <<< nvblox_cpu_gpu_tools [3.66s]                                                                                        
Finished <<< nvblox_isaac_sim [3.61s]                                                                                            
Finished <<< isaac_ros_test [3.82s]
Finished <<< isaac_ros_bi3d_interfaces [14.3s]                                                                                      
Finished <<< isaac_ros_apriltag_interfaces [15.2s]                                                               
Finished <<< realsense2_camera_msgs [15.3s]                                                                   
Starting >>> realsense2_camera
Starting >>> realsense_splitter
Starting >>> realsense2_description
Finished <<< isaac_ros_tensor_list_interfaces [15.6s]                                          
Finished <<< isaac_ros_common [15.7s]                                                          
Finished <<< nvblox_performance_measurement_msgs [15.8s]
Finished <<< nvblox_msgs [16.2s]                                                               
Starting >>> nvblox_nav2
Starting >>> nvblox_rviz_plugin
Finished <<< realsense2_description [2.10s]                                                                            
Finished <<< isaac_ros_visual_slam_interfaces [21.3s]                                                                   
Starting >>> isaac_ros_visual_slam
--- stderr: isaac_ros_visual_slam                                                                        
CMake Error at /opt/ros/humble/install/share/ament_cmake_core/cmake/index/ament_index_get_resource.cmake:62 (message):
  ament_index_get_resource() called with not existing resource ('elbrus'
  'isaac_ros_nitros')
Call Stack (most recent call first):
  CMakeLists.txt:50 (ament_index_get_resource)

---
Failed   <<< isaac_ros_visual_slam [8.06s, exited with code 1]
Aborted  <<< nvblox_nav2 [29.0s]                                                                                    
Aborted  <<< realsense2_camera [35.9s]                                                                               
Aborted  <<< nvblox_rviz_plugin [35.2s]                                                                              
Aborted  <<< realsense_splitter [43.0s]                                                                           
Aborted  <<< nvblox [2min 47s]                                           

Summary: 12 packages finished [2min 48s]
  1 package failed: isaac_ros_visual_slam
  5 packages aborted: nvblox nvblox_nav2 nvblox_rviz_plugin realsense2_camera realsense_splitter
  1 package had stderr output: isaac_ros_visual_slam
  4 packages not processed
admin@ubuntu-Default-string:/workspaces/isaac_ros-dev$ 

I did build successfully before I reinstall the system(x8664, Ubuntu20.04.5LTS ROS2 foxy outside the container), so I used 
git clone --depth=1
to retrive the last commit, but it didn't work either.

Is there anything go wrong?

chivas1000, 2022-10-21:
> **Hi I'm using the VSLAM, but got this error when colcon building:**
>
> admin@ubuntu-Default-string:/workspaces/isaac_ros-dev$ colcon build --symlink-install Starting >>> nvblox_msgs Starting >>> [...]

Oh, it seems that my git command that clone the last commit doesn't work,
is that any way for me to get last commit?

alexmillane, 2022-10-21:
The issue here is that DP2.0 added a dependency on [isaac_ros_nitros](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nitros). 

Please clone this repo and have another go at building.

This additional dependency is not reflected in the installation instructions. Apologies for that. We'll fix it.

chivas1000, 2022-10-22:
> The issue here is that DP2.0 added a dependency on [isaac_ros_nitros](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nitros).
> 
> Please clone this repo and have another go at building.
> 
> This additional dependency is not reflected in the installation instructions. Apologies for that. We'll fix it.

yea after cloning the 
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nitros
build successed
Thank you

jaiveersinghNV, 2022-10-24:
Merged a fix to add the extra `git clone` step [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/pull/55).

closed 2022-10-24 by jaiveersinghNV