NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #58: Is this sufficiently robust for long-term and/or large scale use?

SteveMacenski, 2022-11-11:
If so, I'd be happy to have some documentation about working with Nav2 + VSLAM in our tutorials: https://navigation.ros.org/tutorials/index.html. However, its important that its suitable for Nav2 users for practical applications!

hemalshahNV, 2022-11-15:
We're racking up mileage on Isaac ROS Visual SLAM but if there is feedback on its performance in large-scale use for long-terms, it would be greatly appreciated. It should be suitable for Nav2 users for practical applications but I'd rather have Nav2 users confirm this of course.

SteveMacenski, 2022-11-15:
I don't have any in particular to share, but would love to hear as well as comparison of this to other methods. I'd be happy to have it as a tutorial or integration into ROS 2 as an alternative option for SLAM/localization using vision if its sufficiently good for practical use.

hemalshahNV, 2022-11-16:
Yes, Isaac ROS Visual SLAM does have multi-session mapping (save a map, restore, continue mapping). It would be great to work out a series of benchmarks together to qualify both navigation behavior and subsystems including localization, floor plan mapping, trajectory generation, etc. if there isn't already one you would recommend (continuing the discussion from a ROS Discourse thread).