NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #59: cannot build docker image

Neathle, 2022-11-14:
Following the instructions in the repository to setup the environment, the following error appears:

```
isaac_ros_dev not specified, assuming /home/neathle/workspaces/isaac_ros-dev
Building x86_64.humble.nav2.user base as image: isaac_ros_dev-x86_64 using key x86_64.humble.nav2.user
Using base image name not specified, using ''
Using docker context dir not specified, using Dockerfile directory
Resolved the following Dockerfiles for target image: x86_64.humble.nav2.user
/home/neathle/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.user
/home/neathle/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.x86_64.humble.nav2
Building /home/neathle/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.x86_64.humble.nav2 as image: x86_64-humble-nav2-image with base: 
[+] Building 0.4s (3/3) FINISHED                                                              
 => [internal] load build definition from Dockerfile.x86_64.humble.nav2                  0.1s
 => => transferring dockerfile: 573B                                                     0.0s
 => [internal] load .dockerignore                                                        0.1s
 => => transferring context: 2B                                                          0.0s
 => ERROR [internal] load metadata for nvcr.io/nvidia/isaac/ros:x86_64-humble-nav2_5bd6  0.2s
------
 > [internal] load metadata for nvcr.io/nvidia/isaac/ros:x86_64-humble-nav2_5bd606e569673db8cb1f78f393a5c46b:
------
failed to solve with frontend dockerfile.v0: failed to create LLB definition: failed to authorize: rpc error: code = Unknown desc = failed to fetch anonymous token: unexpected status: 401 Unauthorized
Failed to build base image: isaac_ros_dev-x86_64, aborting.
~/workspaces/isaac_ros-dev/src/isaac_ros_common
```

My system:
* Ubuntu 20.04
* Docker: 20.10.21

**Note:** The log was much longer the first time, but I am unable to remove the cached result and it fails instantly now with the error above.

swapnesh-wani-nvidia, 2022-11-15:
Can you try to get the full log of the error? You can prune the docker image cache and it will start from scratch again.

`docker builder prune`

Neathle, 2022-11-15:
I tried to run the script again today before pruning and It downloaded everything, but now it fails with a different error:

```
isaac_ros_dev not specified, assuming /home/neathle/workspaces/isaac_ros-dev
Building x86_64.humble.nav2.user base as image: isaac_ros_dev-x86_64 using key x86_64.humble.nav2.user
Using base image name not specified, using ''
Using docker context dir not specified, using Dockerfile directory
Resolved the following Dockerfiles for target image: x86_64.humble.nav2.user
/home/neathle/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.user
/home/neathle/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.x86_64.humble.nav2
Building /home/neathle/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.x86_64.humble.nav2 as image: x86_64-humble-nav2-image with base: 
Sending build context to Docker daemon   80.9kB
Step 1/1 : FROM nvcr.io/nvidia/isaac/ros:x86_64-humble-nav2_5bd606e569673db8cb1f78f393a5c46b
x86_64-humble-nav2_5bd606e569673db8cb1f78f393a5c46b: Pulling from nvidia/isaac/ros
3b65ec22a9e9: Pulling fs layer 
fd80d866e8b2: Pulling fs layer 
a364ca75fd6d: Pulling fs layer 
3d4731d03623: Waiting 
53a5c2e0251f: Waiting 
b00ff40d02d9: Waiting 
3036e9b94123: Waiting 
453fdcdda788: Pulling fs layer 
35e12ec5e515: Waiting 
11f61a475a23: Waiting 
24280cf31c9a: Waiting 
79007799e2ed: Waiting 
03eb76abf1e5: Waiting 
4f4fb700ef54: Waiting 
8c80ae8980cd: Waiting 
3918bec26dcf: Waiting 
5988980892dd: Waiting 
f4e902364789: Waiting 
9e630d0e3180: Waiting 
140ee16889c1: Pull complete 
da88e5d79c6f: Pull complete 
f61455589e43: Pull complete 
4363df7b85e8: Pull complete 
6e5e8533bd39: Pull complete 
7435ea83957c: Pull complete 
0dc9580c2401: Pull complete 
15be8ac18616: Pull complete 
2d20876aec99: Pull complete 
b27a26b8bc54: Pull complete 
77e8e1895580: Pull complete 
58486e24d700: Pull complete 
ef9047d558d3: Pull complete 
2020f0da6e65: Pull complete 
e89c028f35a6: Pull complete 
57dc833364e9: Pull complete 
0e8234224d83: Pull complete 
ae4316a64573: Pull complete 
460a2092d34a: Pull complete 
25577f5f0c34: Pull complete 
470a4f1a3861: Pull complete 
c70aab8b800b: Pull complete 
b2c35c86bea9: Pull complete 
b2acdf4d640f: Pull complete 
627909d01b7c: Pull complete 
0a410ef4d0a2: Pull complete 
1e68f15edef6: Pull complete 
ab3a937b2ce7: Pull complete 
b707c805a446: Pull complete 
799b55167afb: Pull complete 
f85fee33b522: Pull complete 
4eac5839291b: Pull complete 
69b60e038821: Pull complete 
5376045668ac: Pull complete 
edff3e3858c0: Pull complete 
be17fae500ce: Pull complete 
b2b1d900c1aa: Pull complete 
82ebd5cb8096: Pull complete 
349cd3bf8b52: Pull complete 
Digest: sha256:feb709e3a48fe8094c7f407ba30a2e7ec67c8f2eb0feb20b6772cf4d2e5ec5ae
Status: Downloaded newer image for nvcr.io/nvidia/isaac/ros:x86_64-humble-nav2_5bd606e569673db8cb1f78f393a5c46b
 ---> 65a4a63cc753
[Warning] One or more build-args [USERNAME USER_GID USER_UID] were not consumed
Successfully built 65a4a63cc753
Successfully tagged x86_64-humble-nav2-image:latest

Use 'docker scan' to run Snyk tests against images to find vulnerabilities and learn how to fix them
Building /home/neathle/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.user as image: isaac_ros_dev-x86_64 with base: x86_64-humble-nav2-image
Sending build context to Docker daemon   80.9kB
Step 1/17 : ARG BASE_IMAGE
Step 2/17 : FROM ${BASE_IMAGE}
 ---> 65a4a63cc753
Step 3/17 : ARG USERNAME=admin
 ---> Running in a9339d0c8f8b
Removing intermediate container a9339d0c8f8b
 ---> dcd29acba51c
Step 4/17 : ARG USER_UID=1000
 ---> Running in fe852eee9500
Removing intermediate container fe852eee9500
 ---> 4179062a641f
Step 5/17 : ARG USER_GID=1000
 ---> Running in 12ea91f0fbb2
Removing intermediate container 12ea91f0fbb2
 ---> e2bfa992630d
Step 6/17 : RUN apt-get update && apt-get install -y         sudo && rm -rf /var/lib/apt/lists/* && apt-get clean
 ---> Running in ba29513a8d16
failed to create shim task: OCI runtime create failed: unable to retrieve OCI runtime error (open /run/containerd/io.containerd.runtime.v2.task/moby/ba29513a8d16700d2c7b3b498ed7be0427c451a74a1568cfad28ee9cfc517e9e/log.json: no such file or directory): exec: "nvidia-container-runtime": executable file not found in $PATH: unknown
Failed to build base image: isaac_ros_dev-x86_64, aborting.
~/workspaces/isaac_ros-dev/src/isaac_ros_common

```

Neathle, 2022-11-16:
I tried to build it again and it worked correctly, I don't know what may have caused it but I am unable to replicate the error. thank you for the help!

closed 2022-11-16 by Neathle

HannahHaensen, 2023-04-24:
ok for me the nvidia forum trick worked as well. mhmm restarting and prune did not do the trick for me! but good to know thanks!