NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #60: How can I evaluate with the KITTI dataset?

Corufa, 2022-11-28:
Currently I was able to try the ros bag file provided with Quickstart.
I want to evaluate with KITTI, but I don't how to handle it.
How can I do a isaac_ros_visual_slam with KITTI?

Thank you!

swapnesh-wani-nvidia, 2022-11-28:
You can do so by creating a ros bag from KITTI images to feed to isaac_ros_visual_slam. Make sure the timestamps are in correct order and are synchronized for left and right images. Also, we have published our results for KITTI benchmark which you can see [here](https://www.cvlibs.net/datasets/kitti/eval_odometry.php?highlight=Elbrus#:~:text=54-,IsaacElbrusGPUSLAM,-0.94%20%25).

Corufa, 2022-11-30:
How didi you create the KITTI ros bag?

Corufa, 2022-12-04:
I could try it with KITTI.

closed 2022-12-04 by Corufa

rocker12121, 2023-09-21:
Hello. Can you please provide more detail on how to run with kitti? or the bag files if possible?

Corufa, 2023-09-25:
Hello. I don't have a ROSbag file because I've already finished the project.