NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #61: "Tracker is lost" on KITTI rosbag

Corufa, 2022-12-04:
I prepared the KITTI dataset in the same way as @DaddyWesker  and ran the sample code. However, I frequently get the message "Tracker is lost". rviz confirmed that there are no sudden jumps in the image sequence.

_Originally posted by @Corufa in https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/issues/43#issuecomment-1336357525_

jaiveersinghNV, 2022-12-06:
Could you please upload and provide the specific ROS bag you're using in your test? We can test it on our end and see if we're able to replicate your problem.

Corufa, 2022-12-06:
You can download my prepared ROS bag [here](https://drive.google.com/drive/folders/1gCOdkzwMQ-Gvi_E9osiWoQFVQOJo2eKs?usp=sharing).
I created a KITTI ROS bag (ROS1 bag) using the same method as @DaddyWesker. Next, I edited the topic name in [rosbag_editor](https://github.com/facontidavide/rosbag_editor).I then converted it to a ROS2 bag using [rosbags](https://gitlab.com/MapIV/rosbags).
[image]

rocker12121, 2023-09-20:
> You can download my prepared ROS bag [here](https://drive.google.com/drive/folders/1gCOdkzwMQ-Gvi_E9osiWoQFVQOJo2eKs?usp=sharing). I created a KITTI ROS bag (ROS1 bag) using the same method as [...]

Hello. Do you have any information about how to make this work on ros2 bags of kitti dataset?

closed 2023-09-21 by Corufa