NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #64: Quickstart doesnt work on Ubuntu 22.04

geoeo, 2022-12-20:
Following the quickstart guide I get the following (related ?) error messages.

When running `./scripts/run_dev.sh` I get:

`docker: Error response from daemon: failed to create shim task: OCI runtime create failed: unable to retrieve OCI runtime error (open /run/containerd/io.containerd.runtime.v2.task/moby/21f1770a617988ea5c6e1cc787167812ac103ea7c8541681c11b0f8b05c87252/log.json: no such file or directory): exec: "nvidia-container-runtime": executable file not found in $PATH: <nil>: unknown.
~/Workspace/isaac_ros-dev/src/isaac_ros_common
`

Ignoring this and trying colcon build I get:
```
CMake Error at CMakeLists.txt:31 (find_package):
  By not providing "Findvpi.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "vpi", but
  CMake did not find one.

  Could not find a package configuration file provided by "vpi" with any of
  the following names:

    vpiConfig.cmake
    vpi-config.cmake

  Add the installation prefix of "vpi" to CMAKE_PREFIX_PATH or set "vpi_DIR"
  to a directory containing one of the above files.  If "vpi" provides a
  separate development package or SDK, be sure it has been installed.

```

hemalshahNV, 2022-12-20:
The first error indicates that you may not have the `nvidia-container-runtime` setup. See [here ](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docs/dev-env-setup.md) on how to set that up. This enables Docker to expose GPUs to your running container. Within the container you'll want to run colcon commands where everything has been setup for you including VPI.

geoeo, 2022-12-21:
The issue was 
1) I didnt have the `nvidia-container-runtime`installed. Since the `deamon.json`was available I missed this. It would be good to explicitly reference https://github.com/NVIDIA/nvidia-container-runtime#installation
2) There are hardcoded paths in the run_dev.sh. i.e. `~/workspaces/..` It would be good to at least mention this in the Readme.md

hemalshahNV, 2022-12-23:
Yes, we'll make these more explicit. `run_dev.sh` has a default path which you can override with an argument, by the way.

jaiveersinghNV, 2023-02-09:
Thanks @geoeo  - we've just merged a PR that explicitly adds a reference to installing the `nvidia-container-runtime` on x86 platforms.
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/pull/55

closed 2023-02-28 by jaiveersinghNV