NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #66: How to connect with my ZED Stereo Camera

Shahrullo, 2022-12-22:
I have tested with basics shown in the tutorial. Now I want to run the Visual_Slam with my ZED Stereo Camera. How can I connect it to the camera and run it. May be it is very basic question, I'm just starting it, that's why don't know much stuff yet!

hemalshahNV, 2022-12-22:
Like with RealSense and other cameras, you'll have to get ZED setup to stream stereo images to the Isaac ROS Visual SLAM node. StereoLabs has documentation on how to set up their SDK/drivers and get their ROS2 packages running (see [here](https://www.stereolabs.com/docs/get-started-with-zed/)).

Shahrullo, 2022-12-23:
@hemalshahNV Thanks for a quick response. Both ROS Visual SLAM and ZED with ROS2 packages are running well separately. I've built Isaac SLAM with docker and ZED-ROS2 locally. Could you provide any idea how to connect ZED with Isaac SLAM? Like setting up inside docker. 

I have no clue how to define `zed2i_camera_node` and putting ZED configurations inside `isaac docker image`

hemalshahNV, 2022-12-23:
You can use run_dev.sh configuration as described [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common#configuring-run_devsh) to add a `Dockerfile.zed` like `Dockerfile.realsense` which sets up the StereoLabs ZED SDK into the environment. `run_dev.sh` will then launch a container which will build your `Dockerfile.zed` layer. You then would add the ZED ROS2 packages to your `/workspaces/isaac_ros-dev` ROS workspace. Finally, update a similar Isaac ROS Visual SLAM launch file to launch the ZED node instead of `realsense2_camera` node, line up the topic names, and you should be up and running.

Shahrullo, 2022-12-28:
@hemalshahNV Got it. I'll try with your guideline accordingly. One more question: Is that possible to create SLAM map with only recorded videos or image ranges? if possible how can I put the video or image data to run with IsaacSLAM?

hemalshahNV, 2023-01-31:
You can record a rosbag from a stereo camera along with the `/camera_info` topic and replay into Isaac ROS Visual SLAM nodes with no issues. We do this often for automated testing.

Shahrullo, 2023-01-31:
@hemalshahNV Okay. Thank you very much!

closed 2023-01-31 by Shahrullo