NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #71: colcon build --symlink-install fails

J1nwenWang, 2023-01-30:
I try to build the project. However, it shows that isaac_ros_test is not finished. Any suggestions to solve this problem? The error logs are shown below.
admin@linux:/workspaces/isaac_ros-dev$ colcon build --symlink-install
Starting >>> isaac_ros_test
Starting >>> isaac_ros_apriltag_interfaces
Starting >>> isaac_ros_common
Starting >>> isaac_ros_tensor_list_interfaces
Starting >>> isaac_ros_visual_slam_interfaces                                                        
Starting >>> isaac_ros_bi3d_interfaces
Finished <<< isaac_ros_common [1.06s]                                                                
Finished <<< isaac_ros_apriltag_interfaces [2.05s]                                                           
Finished <<< isaac_ros_bi3d_interfaces [2.01s]                
Finished <<< isaac_ros_tensor_list_interfaces [2.11s]
Finished <<< isaac_ros_visual_slam_interfaces [2.42s]                                                       
--- stderr: isaac_ros_test                    
Traceback (most recent call last):
  File "setup.py", line 13, in <module>
    setup(
  File "/usr/local/lib/python3.8/dist-packages/setuptools/__init__.py", line 108, in setup
    return distutils.core.setup(**attrs)
  File "/usr/local/lib/python3.8/dist-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
  File "/usr/local/lib/python3.8/dist-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/usr/local/lib/python3.8/dist-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/usr/local/lib/python3.8/dist-packages/setuptools/dist.py", line 1213, in run_command
    super().run_command(command)
  File "/usr/local/lib/python3.8/dist-packages/setuptools/_distutils/dist.py", line 987, in run_command
    cmd_obj.ensure_finalized()
  File "/usr/local/lib/python3.8/dist-packages/setuptools/_distutils/cmd.py", line 111, in ensure_finalized
    self.finalize_options()
  File "/usr/local/lib/python3.8/dist-packages/setuptools/command/develop.py", line 52, in finalize_options
    easy_install.finalize_options(self)
  File "/usr/local/lib/python3.8/dist-packages/setuptools/command/easy_install.py", line 311, in finalize_options
    self.local_index = Environment(self.shadow_path + sys.path)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 1044, in __init__
    self.scan(search_path)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 1077, in scan
    self.add(dist)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 1096, in add
    dists.sort(key=operator.attrgetter('hashcmp'), reverse=True)
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 2631, in hashcmp
    self.parsed_version,
  File "/usr/local/lib/python3.8/dist-packages/pkg_resources/__init__.py", line 2685, in parsed_version
    raise packaging.version.InvalidVersion(f"{str(ex)} {info}") from None
pkg_resources.extern.packaging.version.InvalidVersion: Invalid version: '0.23ubuntu1' (package: distro-info)
---
Failed   <<< isaac_ros_test [2.75s, exited with code 1]

Summary: 5 packages finished [3.16s]
  1 package failed: isaac_ros_test
  1 package had stderr output: isaac_ros_test
  12 packages not processed

NamWoo, 2023-01-30:
this is same for me..

in my case:
* jetson xavier nx dev kit / jetpack 5.0.2

hemalshahNV, 2023-01-30:
We're checking now to see if we can reproduce this locally. Are you running within the Isaac ROS Dev base container or on your own host setup?

J1nwenWang, 2023-01-30:
Thank you very much for the help. I'm running within the Isaac ROS Dev base container.

closed 2023-02-28 by jaiveersinghNV