NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #74: Visual SLAM in Isaac Gym (the Omniverse version)

joehays, 2023-03-03:
Hello,

Is there any support to be able to use the Visual SLAM capabilities from Isaac ROS Visual SLAM in parallel environments in Omniverse Sim/Gym? The ROS2 layer would have to be stripped away as it's not appropriate for the parallel Gym environments used during the RL training. Is this possible?

hemalshahNV, 2023-03-06:
Possible, yes, but it would take some work to strip away the ROS 2 layer and integrate the same underlying libraries with Isaac Gym. Alternatively, you can integrate Isaac Gym and Isaac ROS through your own ROS 2 application.

joehays, 2023-03-08:
ROS is inappropriate when I have over 10k parallel environments running in
Isaac Gym. The power of Isaac Gym is only fully realized when all code runs
on the GPU. Having ROS in the middle removes the primary performance
benefits of Isaac Gym.

closed 2023-08-14 by swapnesh-wani-nvidia