NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #75: is realsense d457 supported on Jetson?

AndreV84, 2023-03-19:
what would be the steps to setup/ run with d457 at Jetson AGX/ Orin ?
Will it work wit the default proposed docker setup steps or the version of the docker container needs to be changed to switch to the version supporting realsense?
Could you comment?

AndreV84, 2023-03-19:
Is it required to address this step separately in order to add realsense camera support `https://github.com/NVIDIA-ISAAC-ROS/.github/blob/main/profile/realsense-setup.md` ? or it will be included in default setup listed in this repository? Thanks

hemalshahNV, 2023-04-17:
You'll likely need an updated [Intel RealSense SDK](https://github.com/IntelRealSense/librealsense) probably to recognize such a new camera as the D457. We have no plans to test with this camera at present but nothing preventing you from getting it to work.

closed 2023-08-14 by swapnesh-wani-nvidia