NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #77: the thing just dont't make sense

FPSychotic, 2023-04-01:
Orin NX use JP5.1,but there is not JP5.1 for it, ORIN use 20.04, so it use Foxy but Issac-ros use Humble in docker, docker is not given in Orin and Humble is not for 20.04, Docker is not given in Orin 20.04 image, I don't want use docker,I cannot use docker, I buy a linux board not a docker board, you make software for docker users , not for jetson users, you are nVidia not docker, so why make a docker board,

It doesn't make sense, from first day of ROS gems was impossible to use, in any update was impossible to use , still is impossible to use. Just bin the whole thing, I mean the whole thing and if you can the whole nvidia software ecosystem.

hemalshahNV, 2023-04-01:
Thank you for your writing up some of your frustrations here. Let me address some of it below.

> Orin NX use JP5.1,but there is not JP5.1 for it
Jetpack 5.1.1 supports Jetson Orin NX 8GB [here](https://developer.nvidia.com/embedded/jetpack). Perhaps this was not available when you last checked.

> ORIN use 20.04, so it use Foxy
> Humble is not for 20.04
Jetpack 5.x is based on Ubuntu 20.04 which is a Tier 1 platform for ROS 2 Foxy (meaning, OSRF builds the binaries for it). It is a Tier 3 platform for ROS 2 Humble, however (meaning, OSRF maintains source compatibility but does not build the binaries for it).
>  Issac-ros use Humble in docker
We build ROS 2 Humble and required packages from source for you along with other configured libraries delivered as a Docker images to help solve the above situation ([Isaac ROS Dev Base](https://catalog.ngc.nvidia.com/orgs/nvidia/teams/isaac/containers/ros)). Would a ROS 2 Humble debian package to install on Ubuntu 20.04 be helpful? Do you leverage any DNN inference packages such as isaac_ros_tensor_rt or isaac_ros_triton?

> docker is not given in Orin
Docker should be available in JetPack 5.x but perhaps I misunderstood this.

> I don't want use docker,I cannot use docker
I am curious why but I get that's not your point. You can follow the command-lines in the Dockerfiles themselves and prepare your own system. We're exploring supporting a bare metal setup but that does require users to do a lot more than just a "docker run".

> I buy a linux board not a docker board, you make software for docker users , not for jetson users, you are nVidia not docker, so why make a docker board,
Besides the point here, but containerization is based on a key Linux technology of "chroot jail" providing a "lightweight" VM (share the same host kernel but partitioned environment otherwise). We provide our developer experience through Docker images for ease-of-use for Jetson users.

> It doesn't make sense, from first day of ROS gems was impossible to use, in any update was impossible to use , still is impossible to use. 
I take it that we provide Docker images to setup your environment and you don't want to use Docker, hence Isaac ROS gems are impossible to use?

>  Just bin the whole thing, I mean the whole thing and if you can the whole nvidia software ecosystem.
We're always looking for ways to make the NVIDIA software ecosystem easier for our users. We can't do it all at once but we'll keep improving to make this all better for everyone.

closed 2023-04-24 by FPSychotic

FPSychotic, 2023-04-24:
No, it doesn't work with docker too. I oppened the issue.

> I am curious why but I get that's not your point. You can follow the command-lines in the Dockerfiles themselves and prepare your own system

No, you cannot, again will fail to build in nvidia Orin NX

> Besides the point here, but containerization is based on a key Linux technology of "chroot jail" providing a "lightweight" VM (share the same host kernel but partitioned environment otherwise). We provide our developer experience through Docker images for ease-of-use for Jetson users.

Easy is relative. As convenient, please allow me decide that myself, how is of convenient something that doesn't work and I need wait to someone upload a docker image with the fix when I could do it my self, or need take the limitations , bugs , time to learn of something that I didn't choose, nvidia chosen for me?, the native way of work in linux is native, and is what give you the control, that is what any company wants, if I want use docker, a VM a cloud I can do it by myself, I dont need nVidia chose a