NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #82: Failed to build base image: isaac_ros_dev-aarch64, aborting.

FPSychotic, 2023-04-24:
```
nrover@nrover:~/workspaces/isaac_ros-dev/src$ git clone https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam
Cloning into 'isaac_ros_visual_slam'...
remote: Enumerating objects: 359, done.
remote: Counting objects: 100% (84/84), done.
remote: Compressing objects: 100% (63/63), done.
remote: Total 359 (delta 26), reused 50 (delta 20), pack-reused 275
Receiving objects: 100% (359/359), 199.30 KiB | 857.00 KiB/s, done.
Resolving deltas: 100% (145/145), done.
Filtering content: 100% (29/29), 332.96 MiB | 3.35 MiB/s, done.
nrover@nrover:~/workspaces/isaac_ros-dev/src$ git clone https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common
Cloning into 'isaac_ros_common'...
remote: Enumerating objects: 758, done.
remote: Counting objects: 100% (162/162), done.
remote: Compressing objects: 100% (110/110), done.
remote: Total 758 (delta 64), reused 79 (delta 47), pack-reused 596
Receiving objects: 100% (758/758), 451.25 KiB | 1.50 MiB/s, done.
Resolving deltas: 100% (268/268), done.
Filtering content: 100% (8/8), 806.54 KiB | 627.00 KiB/s, done.
nrover@nrover:~/workspaces/isaac_ros-dev/src$ git clone https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nitros
Cloning into 'isaac_ros_nitros'...
remote: Enumerating objects: 1377, done.
remote: Counting objects: 100% (1377/1377), done.
remote: Compressing objects: 100% (892/892), done.
remote: Total 1377 (delta 448), reused 1262 (delta 342), pack-reused 0
Receiving objects: 100% (1377/1377), 607.87 KiB | 1.53 MiB/s, done.
Resolving deltas: 100% (448/448), done.
Filtering content: 100% (82/82), 229.49 MiB | 3.60 MiB/s, done.
nrover@nrover:~/workspaces/isaac_ros-dev/src$ cd ${ISAAC_ROS_WS}/src/isaac_ros_common && \
>   ./scripts/run_dev.sh ${ISAAC_ROS_WS}
Error: Failed to call git rev-parse --git-dir: exit status 128 
Building aarch64.ros2_humble.user base as image: isaac_ros_dev-aarch64 using key aarch64.ros2_humble.user
Using base image name not specified, using ''
Using docker context dir not specified, using Dockerfile directory
Resolved the following Dockerfiles for target image: aarch64.ros2_humble.user
/home/nrover/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.user
/home/nrover/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.aarch64.ros2_humble
Building /home/nrover/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.aarch64.ros2_humble as image: aarch64-ros2_humble-image with base: 
[+] Building 845.6s (4/4) FINISHED                                              
 => [internal] load build definition from Dockerfile.aarch64.ros2_humble   0.0s
 => => transferring dockerfile: 575B                                       0.0s
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => [internal] load metadata for nvcr.io/nvidia/isaac/ros:aarch64-ros2_hu  3.9s
 => ERROR [1/1] FROM nvcr.io/nvidia/isaac/ros:aarch64-ros2_humble_c8ad2  841.7s
 => => resolve nvcr.io/nvidia/isaac/ros:aarch64-ros2_humble_c8ad2c6ecc687  0.0s
 => => sha256:a8b1b944a2887ae21de58136aac7c36068913c475 23.66kB / 23.66kB  0.0s
 => => sha256:0a1def070f0bd005945e0ead69f08e488bbc0 234.18MB / 234.18MB  165.2s
 => => sha256:f0d41bdab50f2b735b49f10b946cfcf6b6482f5775e 4.25kB / 4.25kB  1.8s
 => => sha256:78723028f99f1d36e23952b4c8134a997ab8c5db464 9.78kB / 9.78kB  0.0s
 => => sha256:f04b4bbe15805316c8fda79beedd3b77e6b1ffcd 27.19MB / 27.19MB  16.3s
 => => sha256:6328a89a71f9e593c89369d44599854fdb241ec009e 4.14kB / 4.14kB  3.1s
 => => sha256:f7adbdc55099abf403d0f5f641e6b4adf565717a5172f44 900B / 900B  4.3s
 => => sha256:4d9a358decc4b98b779e4af5d96a9193a48fb1bd18f3646 205B / 205B  6.0s
 => => sha256:3dfc1c856868daa08ed3e7fc42e65c9901389e20752f57c 210B / 210B  7.6s
 => => sha256:ef8e5ac0e0fe7bbb56f45b5fbddf88c57535ec1859417cf 290B / 290B  8.7s
 => => sha256:acffb039d986ce986cdac87bc14867357361865b03ae6ec 298B / 298B  9.8s
 => => sha256:cb612f8e41f48356a6320d326a3dc34da98ff154 14.49kB / 14.49kB  10.6s
 => => sha256:964b1c4825f38a0e4f31032fe0cec60f5db20fcb 95.97MB / 95.97MB  78.8s
 => => extracting sha256:f04b4bbe15805316c8fda79beedd3b77e6b1ffcd0acf8122  0.5s
 => => sha256:65f859c93f73608bdc5389b1227a4233c44c412b 18.27MB / 18.27MB  26.0s
 => => sha256:a084f32754a1632420a462ca658a72d66366ea11fb 2.19kB / 2.19kB  27.5s
 => => sha256:5ccd4a259fa8c7f3ac08c338527544275cb5ed9e 13.78MB / 13.78MB  40.0s
 => => sha256:67964a03da1d786482fa48f39762dc4d1c712a33 13.69MB / 13.69MB  51.1s
 => => sha256:e93e45fba8a1609d2859caa55afc347b48ce9 639.23MB / 639.23MB  554.0s
 => => sha256:30f86aa4b0e9395b3182a6ad84384e7364de5be 28.60MB / 28.60MB  110.2s
 => => sha256:06e404a62dc0081bd6686c738e216983f1eef093f 4.88MB / 4.88MB  117.3s
 => => sha256:6f0ad9fb03443b090c0300973bb9d16c2c36474 22.13MB / 22.13MB  134.3s
 => => sha256:dc1a845923ddf836408a99594d8c23b6cb493d207 5.52MB / 5.52MB  137.9s
 => => sha256:5a581758e63639f0b54cce331fd7aeb010e2c6ba8 1.00GB / 3.55GB  841.7s
[... 33 lines cut ...]
Failed to build base image: isaac_ros_dev-aarch64, aborting.
~/workspaces/isaac_ros-dev/src/isaac_ros_common
nrover@nrover:~/workspaces/isaac_ros-dev/src/isaac_ros_common$ cd ${ISAAC_ROS_WS}/src/isaac_ros_common &&   ./scripts/run_dev.sh ${ISAAC_ROS_WS}
Error: Failed to call git rev-parse --git-dir: exit status 128 
Building aarch64.ros2_humble.user base as image: isaac_ros_dev-aarch64 using key aarch64.ros2_humble.user
Using base image name not specified, using ''
Using docker context dir not specified, using Dockerfile directory
Resolved the following Dockerfiles for target image: aarch64.ros2_humble.user
/home/nrover/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.user
/home/nrover/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.aarch64.ros2_humble
Building /home/nrover/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.aarch64.ros2_humble as image: aarch64-ros2_humble-image with base: 
[+] Building 0.6s (3/3) FINISHED                                                
 => [internal] load build definition from Dockerfile.aarch64.ros2_humble   0.0s
 => => transferring dockerfile: 58B                                        0.0s
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => ERROR [internal] load metadata for nvcr.io/nvidia/isaac/ros:aarch64-r  0.5s
------
 > [internal] load metadata for nvcr.io/nvidia/isaac/ros:aarch64-ros2_humble_c8ad2c6ecc68747eca2f1df62e25ae5e:
------
failed to solve with frontend dockerfile.v0: failed to create LLB definition: failed to authorize: rpc error: code = Unknown desc = failed to fetch anonymous token: unexpected status: 401 Unauthorized
Failed to build base image: isaac_ros_dev-aarch64, aborting.
~/workspaces/isaac_ros-dev/src/isaac_ros_common

```

FPSychotic, 2023-04-24:
second try...
different error.
```
=> sha256:5a581758e63639f0b54cce331fd7aeb010e2c6ba88bfc7c25e322f91bb36ec8f 652.21MB / 3.55GB                             864.3s
 => => extracting sha256:0a1def070f0bd005945e0ead69f08e488bbc02a6c8911eb5532a27a020aef1e4                                      4.7s
 => => sha256:53784718ebe6f43e11c6a24f152360d7606b748f0fd337108cd25662515ac807 35.95kB / 35.95kB                             204.5s
 => => sha256:fe0dcfb81a1d8f824fe19ed339b194cbe348aa202467bed6ea2cd051a2e745f3 37.80kB / 37.80kB                             205.9s
 => => sha256:bef4be2e0bcaf94c756a524c6bc3fdb7275ff5bc56484bc8435e3be5a9372599 240.73MB / 240.73MB                           426.0s
 => => extracting sha256:f0d41bdab50f2b735b49f10b946cfcf6b6482f5775e08afdc138b1cc27a1ce91                                      0.0s
 => => extracting sha256:6328a89a71f9e593c89369d44599854fdb241ec009e775cec3b3cdb7108af8a0                                      0.0s
 => => extracting sha256:f7adbdc55099abf403d0f5f641e6b4adf565717a5172f44ebe69bd27c987cb71                                      0.0s
 => => extracting sha256:4d9a358decc4b98b779e4af5d96a9193a48fb1bd18f3646b4a436cea1d0b0bda                                      0.0s
 => => extracting sha256:3dfc1c856868daa08ed3e7fc42e65c9901389e20752f57c3089524341c36e24a                                      0.0s
 => => extracting sha256:ef8e5ac0e0fe7bbb56f45b5fbddf88c57535ec1859417cf382c63908be240fd1                                      0.0s
 => => extracting sha256:acffb039d986ce986cdac87bc14867357361865b03ae6ec4a88d0b8995d43efa                                      0.0s
 => => extracting sha256:cb612f8e41f48356a6320d326a3dc34da98ff1547949e7ce031ae24679b8a8b6                                      0.0s
 => => extracting sha256:964b1c4825f38a0e4f31032fe0cec60f5db20fcb287a4db670108b4e46cf3dcf                                      1.6s
 => => extracting sha256:65f859c93f73608bdc5389b1227a4233c44c412b5cf1e94936f182904b974d4f                                      0.5s
 => => extracting sha256:a084f32754a1632420a462ca658a72d66366ea11fba39d4d8fac49c28854ad0c                                      0.0s
 => => extracting sha256:5ccd4a259fa8c7f3ac08c338527544275cb5ed9e37f1be4fcc274e3e03de990c                                      0.4s
 => => extracting sha256:67964a03da1d786482fa48f39762dc4d1c712a33df82bdf8734160af5177ad60                                      0.2s
 => => sha256:9708e8630251c05083d1d78be2683b1861bfb338748c7fbe43cb254836eae65b 23.36MB / 23.36MB                             453.1s
 => => sha256:9682d7c8965dce3939a4e91892a7f1f4bd14a03382b94bd26886e60eee093f53 423.62MB / 1.18GB                             864.3s
 => => sha256:ee2b30cf1b123cfd622f46043bd99d306b32a5a1a5ef75a15b3d962ad80f30bd 0B / 2.15MB                                   864.3s
 => => extracting sha256:e93e45fba8a1609d2859caa55afc347b48ce904ca855cb5817b3d92141d20b79                                     12.5s
 => => extracting sha256:30f86aa4b0e9395b3182a6ad84384e7364de5be8ee86551e7ab5d2997152ef9a                                      0.5s
 => => extracting sha256:06e404a62dc0081bd6686c738e216983f1eef093f8282e553c8e61f0967063ea                                      0.1s
 => => extracting sha256:6f0ad9fb03443b090c0300973bb9d16c2c36474c00450f70092ad684d0000f41                                      0.4s
 => => extracting sha256:dc1a845923ddf836408a99594d8c23b6cb493d207b195bdb5e7c94a290d2dc9d                                      0.1s
------
 > [1/1] FROM nvcr.io/nvidia/isaac/ros:aarch64-ros2_humble_c8ad2c6ecc68747eca2f1df62e25ae5e@sha256:78723028f99f1d36e23952b4c8134a997ab8c5db464a43984138c7d120aa5896:
------
failed to copy: httpReadSeeker: failed open: failed to authorize: rpc error: code = Unknown desc = failed to fetch anonymous token: unexpected status: 401 Unauthorized
Failed to build base image: isaac_ros_dev-aarch64, aborting.

```

hemalshahNV, 2023-04-24:
Thanks for posting this. We're looking into it. The issue is "failed to fetch anonymous token: unexpected status: 401 Unauthorized" which means docker registry is requiring us to log in before pulling the images but you should not need to. You can log into NGC and run docker login with your NGC token from [setup](https://ngc.nvidia.com/setup). As a workaround, you can also build the image locally by deleting the file "Dockerfile.aarch64.ros2_humble" which will then cause the scripts to build from scratch rather than use the prebuilt image (note: this will take a while).

ikhann, 2023-04-30:
Hi [FPSychotic](https://github.com/FPSychotic)

You can temporarily fix it with assigning `DOCKER_BUILDKIT=0`
here https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/scripts/build_base_image.sh#L71

It works for me.

closed 2023-08-14 by swapnesh-wani-nvidia

mmazzarino, 2023-10-09:
I have a similar issue. Below is part of the output:

`
#!/bin/bash

# Repositório principal - ISAAC ROS VISUAL SLAM
# https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam

# Tutorial de referência deste script (Visual SLAM usando uma câmera RealSense com IMU integrada)
# https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/main/docs/tutorial-realsense.md

# Configuração do ambiente de desenvolvimento Isaac ROS
# https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docs/dev-env-setup.md

# Configuração do Isaac ROS RealSense
# https://github.com/NVIDIA-ISAAC-ROS/.github/blob/main/profile/realsense-setup.md

cd ${ISAAC_ROS_WS}

if [ ! -d "src" ]; then
  mkdir src
fi

cd /tmp && \
if [ ! -d "librealsense" ]; then
  git clone https://github.com/IntelRealSense/librealsense && \
  cd librealsense && \
  git lfs pull && \
  cd ..
fi
cd librealsense && \
./scripts/setup_udev_rules.sh && \

cd ${ISAAC_ROS_WS}/src

if [ ! -d "isaac_ros_common" ]; then
  git clone https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common && \
  cd isaac_ros_common && \
  git checkout release-dp3.1 && \
  git lfs pull &&\
  cd ..
fi

if [ ! -d "isaac_ros_visual_slam" ]; then
  git clone https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam && \
  cd isaac_ros_visual_slam && \
  git checkout release-dp3.1 && \
  git lfs pull &&\
  cd ..
fi

if [ ! -d "isaac_ros_nitros" ]; then
  git clone https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nitros && \
  cd isaac_ros_nitros && \
  git checkout release-dp3.1 && \
  git lfs pull &&\
  cd ..
fi

if [ ! -d "realsense-ros" ]; then
    git clone https://github.com/IntelRealSense/realsense-ros.git -b 4.51.1 && \
    cd realsense-ros && \
    git lfs pull && \
    cd ..
fi

cd ${ISAAC_ROS_WS}/src/isaac_ros_common/scripts && \
  touch .isaac_ros_common-config && \
  echo CONFIG_IMAGE_KEY=ros2_humble.realsense > .isaac_ros_common-config

cd ${ISAAC_ROS_WS}/src

read -p "Pressione Enter depois de conectar a camera RealSense" 

cd ${ISAAC_ROS_WS}/src/isaac_ros_common && \
  ./scripts/run_dev.sh ${ISAAC_ROS_WS}

....
...
..
.
`

Building aarch64.ros2_humble.realsense.user base as image: isaac_ros_dev-aarch64 using key aarch64.ros2_humble.realsense.user
Using base image name not specified, using ''
Using docker context dir not specified, using Dockerfile directory
WARNING: Plugin "/home/mmazza/.docker/cli-plugins/docker-buildx" is not valid: failed to fetch metadata: fork/exec /home/mmazza/.docker/cli-plugins/docker-buildx: exec format error
Resolved the following Dockerfiles for target image: aarch64.ros2_humble.realsense.user
/home/mmazza/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.user
/home/mmazza/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.realsense
/home/mmazza/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.aarch64.ros2_humble
Building /home/mmazza/workspaces/isaac_ros-dev/src/isaac_ros_common/scripts/../docker/Dockerfile.aarch64.ros2_humble as image: aarch64-ros2_humble-image with base:
failed to fetch metadata: fork/exec /home/mmazza/.docker/cli-plugins/docker-buildx: exec format error

ERROR: BuildKit is enabled but the buildx component is missing or broken.
       Install the buildx component to build images with BuildKit:
       https://docs.docker.com/go/buildx/
Failed to build base image: isaac_ros_dev-aarch64, aborting.