NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #84: Question about ISAAC SIm and SLAM

HannahHaensen, 2023-04-26:
Hey thanks for providing the tutorial

I downloaded the scene and everything: ../../../Environments/Simple_Warehouse/warehouse_with_forklifts.usd

but it can not find that path also I checked: https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/reference_assets.html

but i don't have an Isaac folder neither on the server or on the local machine

could you help me?

HannahHaensen, 2023-04-26:
The files are at a new location: omniverse://localhost/NVIDIA/Assets/Isaac/2022.2.1/Isaac//Environments/Simple_Warehouse/Stage/warehouse_extras.usd

https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/install_workstation.html#isaac-sim-setup-native-main

jaiveersinghNV, 2023-05-03:
It sounds like this might be an issue with [Isaac Sim](https://developer.nvidia.com/isaac-sim), not [Isaac ROS](https://developer.nvidia.com/isaac-ros).

Could you please start a thread detailing your issue on the Isaac Sim Developer Forum [here](https://forums.developer.nvidia.com/c/omniverse/simulation/69)?

HannahHaensen, 2023-05-04:
Thanks I contacted the forum!

closed 2023-05-04 by HannahHaensen