NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #87: How to visualize map data(data.mdb) from SaveMap ROS 2 Action?

mat-macnica, 2023-05-15:
Hello, 

We got a map data file "data.mdb" by following below link;
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/main/docs/tutorial-isaac-sim.md#saving-and-using-the-map
  ros2 action send_goal /visual_slam/save_map isaac_ros_visual_slam_interfaces/action/SaveMap "{map_url: /path/to/save/the/map}"

Question:
 Are there any recommended way to visualize the "data.mdb" map file by using visualizing tool such as Rviz2?

--
Best regards,

jaiveersinghNV, 2023-05-15:
There isn't a way to visualize the `data.mdb` file directly, but it can be loaded back into Isaac ROS Visual SLAM via the `isaac_ros_visual_slam_interfaces/action/LoadMapAndLocalize` Action and then visualized in RViz, as shown [here](isaac_ros_visual_slam_interfaces/action/LoadMapAndLocalize).

mat-macnica, 2023-05-16:
Hello,
 
Thank you very much for prompt reply.
I really appreciate your help.
I'd like to check the behavior from your guide.

--
Best regards,

closed 2023-06-05 by jaiveersinghNV