NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #91: Using  isaac_ros_visual_slam on a UAV

mzahana, 2023-05-30:
Hi!
I have the following setup 
* Intel Realsense D455 camera (firmware is updated), and using `realsense-ros` `4.51.1`
* Jetson Xavier NX with SSD and with 8GB RAM at mode (20W 6 CORE)
* A quadcopter 
* The D455 is rigidly mounted on the quadcopter and facing front 
* I need to get the depth and color images for some object detection and localization, in addition to the infra1/2 for VSLAM

I managed to run the VSLAM with realsense on Jetson Xavier NX. With relatively slow motion (quadcopter is not flying, but I move it by hand for testing), the odom is fine. When I start to move the quadcopter (by hand) faster, I start to get `Tracker lost` frequently although I am moving in feature-rich environment .

So, what could be the issue, and how to solve it to get reliable odometry?

Thank you.

swapnesh-wani-nvidia, 2023-08-14:
I am not sure if this is a problem anymore, however, it looks like when you are moving your uav faster with your hand, it causes motion blur which will result in `Tracker is lost`. Additionally, there are frame "drops" (warning messages showing that input images to Isaac ROS Visual Slam are not steady) and this will also result in bad tracking.

mzahana, 2023-08-21:
Hi @swapnesh-wani-nvidia 
Thanks for your reply. 
I have replaced the Jetson Xavier NX with Orin NX. I am suing the same camera D455. Here is the output

```
ros2 launch isaac_ros_visual_slam isaac_ros_visual_slam_realsense.launch.py 
[INFO] [launch]: All log files can be found below /home/admin/.ros/log/2023-08-21-07-13-37-588341-orin-nx-31595
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [31606]
[INFO] [realsense2_camera_node-2]: process started with pid [31608]
[component_container-1] [INFO] [1692602018.195577355] [visual_slam_launch_container]: Load Library: /workspaces/isaac_ros-dev/install/isaac_ros_visual_slam/lib/libvisual_slam_node.so
[realsense2_camera_node-2] [INFO] [1692602018.210915211] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-2] [INFO] [1692602018.211124594] [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-2] [INFO] [1692602018.211154387] [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-2] [INFO] [1692602018.466795712] [camera.camera]: Device with serial number 207322250752 was found.
[realsense2_camera_node-2] 
[realsense2_camera_node-2] [INFO] [1692602018.466957222] [camera.camera]: Device with physical ID 2-3-5 was found.
[realsense2_camera_node-2] [INFO] [1692602018.466980903] [camera.camera]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-2] [INFO] [1692602018.467621340] [camera.camera]: Device with port number 2-3 was found.
[realsense2_camera_node-2] [INFO] [1692602018.467672190] [camera.camera]: Device USB type: 3.2
[component_container-1] [INFO] [1692602018.476514724] [visual_slam_launch_container]: Found class: rclcpp_components::NodeFactoryTemplate<isaac_ros::visual_slam::VisualSlamNode>
[component_container-1] [INFO] [1692602018.476636169] [visual_slam_launch_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<isaac_ros::visual_slam::VisualSlamNode>
[realsense2_camera_node-2] [INFO] [1692602018.478754799] [camera.camera]: getParameters...
[realsense2_camera_node-2] [INFO] [1692602018.483103328] [camera.camera]: JSON file is not provided
[realsense2_camera_node-2] [INFO] [1692602018.484236806] [camera.camera]: Device Name: Intel RealSense D455
[realsense2_camera_node-2] [INFO] [1692602018.484276551] [camera.camera]: Device Serial No: 207322250752
[realsense2_camera_node-2] [INFO] [1692602018.484305608] [camera.camera]: Device physical port: 2-3-5
[realsense2_camera_node-2] [INFO] [1692602018.484321289] [camera.camera]: Device FW version: 5.14.0
[realsense2_camera_node-2] [INFO] [1692602018.484350314] [camera.camera]: Device Product ID: 0x0B5C
[realsense2_camera_node-2] [INFO] [1692602018.484363114] [camera.camera]: Sync Mode: Off
[component_container-1] [INFO] [1692602018.542016013] [visual_slam_node]: cuVSLAM version: 11.0
[realsense2_camera_node-2] [WARN] [1692602022.427257981] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-2] [WARN] [1692602023.343521644] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-2] [INFO] [1692602023.457865926] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-2] [INFO] [1692602023.899850255] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-2] [INFO] [1692602024.128711399] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 640, Height: 360, FPS: 90
[realsense2_camera_node-2] [INFO] [1692602024.128861036] [camera.camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 640, Height: 360, FPS: 90
[realsense2_camera_node-2] [INFO] [1692602024.128989136] [camera.camera]: Stopping Sensor: Motion Module
[realsense2_camera_node-2] [INFO] [1692602024.148161229] [camera.camera]: Starting Sensor: Motion Module
[realsense2_camera_node-2] [INFO] [1692602024.232241816] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 200
[realsense2_camera_node-2] [INFO] [1692602024.233334012] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[realsense2_camera_node-2] [WARN] [1692602024.242075038] [camera.camera]: 
[realsense2_camera_node-2] [INFO] [1692602024.254367767] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-2] [WARN] [1692602024.904798919] [camera.camera]: XXX Hardware Notification:Motion Module failure,1.6926e+12,Error,Hardware Error
[realsense2_camera_node-2] [WARN] [1692602024.904963116] [camera.camera]: Hardware Notification:Motion Module failure,1.6926e+12,Error,Hardware Error
[component_container-1] [INFO] [1692602035.804148538] [visual_slam_node]: Time taken by CUVSLAM_WarmUpGPU(): 17.261979
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/visual_slam_node' in container '/visual_slam_launch_container'
[component_container-1] [INFO] [1692602035.816379790] [visual_slam_node]: left_pose_right reading from CameraInfo
[component_container-1] [INFO] [1692602035.816497778] [visual_slam_node]: Baseline is : 0.095124
[component_container-1] [INFO] [1692602035.816527315] [visual_slam_node]: left_camera_pose_imu reading from TF Tree
[component_container-1] [INFO] [1692602035.816700185] [visual_slam_node]: Use use_gpu: true
[component_container-1] [INFO] [1692602035.816723481] [visual_slam_node]: Enable IMU Fusion: true
[component_container-1] [INFO] [1692602035.816735290] [visual_slam_node]: gyroscope_noise_density: 0.000244
[... cut ...]
[component_container-1] [WARN] [1692602122.034775281] [visual_slam_node]: Delta between current and previous frame [300.884480] is above threshold [22.000000]
[component_container-1] [WARN] [1692602122.274707829] [visual_slam_node]: Delta between current and previous frame [300.019456] is above threshold [22.000000]
```

There is a lot of jittering, and I am not sure why. I am using the same launch file for realsense camera.

This happens with the camera hand-held and with slow speed.

I am not sure how to fix this issue.

swapnesh-wani-nvidia, 2023-08-21:
Could you check if the o/p from RealSense node is coming at the desired rate of 90 fps?
`ros2 topic hz --window 100 <image_topic_name>`

mzahana, 2023-08-22:
Hi @swapnesh-wani-nvidia 

Here is the output.

```bash
ros2 topic hz --window 100 /camera/infra1/image_rect_raw
WARNING: topic [/camera/infra1/image_rect_raw] does not appear to be published yet
average rate: 67.906
	min: 0.001s max: 0.032s std dev: 0.01376s window: 68
average rate: 75.018
	min: 0.001s max: 0.033s std dev: 0.01436s window: 100
average rate: 66.711
	min: 0.001s max: 0.033s std dev: 0.01518s window: 100
average rate: 46.834
	min: 0.001s max: 0.035s std dev: 0.01507s window: 100
average rate: 42.833
	min: 0.001s max: 0.035s std dev: 0.01444s window: 100
average rate: 78.792
	min: 0.001s max: 0.035s std dev: 0.01494s window: 100
average rate: 88.386
	min: 0.001s max: 0.034s std dev: 0.01432s window: 100
average rate: 88.377
	min: 0.001s max: 0.035s std dev: 0.01432s window: 100
average rate: 88.419
	min: 0.001s max: 0.033s std dev: 0.01445s window: 100
average rate: 88.282
	min: 0.001s max: 0.033s std dev: 0.01440s window: 100
average rate: 90.837
	min: 0.001s max: 0.033s std dev: 0.01425s window: 100
average rate: 90.711
	min: 0.001s max: 0.032s std dev: 0.01423s window: 100
average rate: 88.261
	min: 0.001s max: 0.033s std dev: 0.01423s window: 100
average rate: 90.780
	min: 0.001s max: 0.032s std dev: 0.01416s window: 100
average rate: 88.235
	min: 0.001s max: 0.033s std dev: 0.01430s window: 100
average rate: 90.883
	min: 0.001s max: 0.035s std dev: 0.01436s window: 100
average rate: 88.161
	min: 0.001s max: 0.036s std dev: 0.01465s window: 100
average rate: 88.210
	min: 0.001s max: 0.034s std dev: 0.01430s window: 100
average rate: 90.835
	min: 0.001s max: 0.034s std dev: 0.01436s window: 100
average rate: 90.856
	min: 0.001s max: 0.034s std dev: 0.01432s window: 100
average rate: 90.845
	min: 0.001s max: 0.035s std dev: 0.01422s window: 100
average rate: 90.786
	min: 0.001s max: 0.033s std dev: 0.01434s window: 100
average rate: 90.832
	min: 0.001s max: 0.032s std dev: 0.01438s window: 100
average rate: 88.207
	min: 0.001s max: 0.033s std dev: 0.01454s window: 100
average rate: 90.764
	min: 0.001s max: 0.033s std dev: 0.01428s window: 100
```

for the other image
```bash
ros2 topic hz --window 100 /camera/infra2/image_rect_raw
WARNING: topic [/camera/infra2/image_rect_raw] does not appear to be published yet
average rate: 38.718
	min: 0.001s max: 0.035s std dev: 0.01331s window: 40
average rate: 56.445
	min: 0.001s max: 0.035s std dev: 0.01567s window: 100
average rate: 63.738
	min: 0.001s max: 0.035s std dev: 0.01540s window: 100
average rate: 73.341
	min: 0.001s max: 0.035s std dev: 0.01479s window: 100
average rate: 88.232
	min: 0.001s max: 0.032s std dev: 0.01420s window: 100
average rate: 88.149
	min: 0.001s max: 0.033s std dev: 0.01438s window: 100
average rate: 88.376
	min: 0.001s max: 0.033s std dev: 0.01436s window: 100
average rate: 90.783
	min: 0.001s max: 0.032s std dev: 0.01418s window: 100
average rate: 88.230
	min: 0.001s max: 0.033s std dev: 0.01425s window: 100
average rate: 90.875
	min: 0.000s max: 0.035s std dev: 0.01423s window: 100
average rate: 88.177
	min: 0.000s max: 0.035s std dev: 0.01445s window: 100
average rate: 90.918
	min: 0.001s max: 0.034s std dev: 0.01424s window: 100
average rate: 88.111
	min: 0.001s max: 0.033s std dev: 0.01439s window: 100
average rate: 90.784
	min: 0.001s max: 0.035s std dev: 0.01428s window: 100
average rate: 88.511
	min: 0.001s max: 0.035s std dev: 0.01441s window: 100
average rate: 88.249
	min: 0.001s max: 0.033s std dev: 0.01445s window: 100
average rate: 88.294
	min: 0.001s max: 0.033s std dev: 0.01440s window: 100
average rate: 88.365
	min: 0.001s max: 0.033s std dev: 0.01432s window: 100
average rate: 90.722
	min: 0.001s max: 0.033s std dev: 0.01421s window: 100
average rate: 90.834
	min: 0.000s max: 0.034s std dev: 0.01428s window: 100
average rate: 90.809
	min: 0.001s max: 0.034s std dev: 0.01434s window: 100
```

Please note that I am also streaming RGB and depth images as I need them, and I am using USB C cable.

furkansariyildiz, 2023-09-22:
Hello,

Do you have any solution in this state?

swapnesh-wani-nvidia, 2023-10-03:
@mzahana, could you try with depth stream off along with emitter off? Leaving the IR projector on could result in "Tracker Lost" output. See the [troubleshooting section](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam#troubleshooting-suggestions) 5(iv). I understand that you need depth images for other things and I am suggesting turning off the depth images to find the root cause.

mzahana, 2023-10-03:
@swapnesh-wani-nvidia thanks for your reply. I already did that, and the issue is still there.