NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #95: How to improve performance on Jetson Xavier NX?

fxprime, 2023-06-10:
Hi. Thank you for great visual slam.

I begin with confuse about frame orientation. But now i can manage to run it properly(atleast i guess)
The problem is, while it run with rectified images from machine transport to docker that isaac workspace is located, it seem not smooth as expect. So i record a bag file below then move to docker folder. But it improve just a little bit. The problem still occur with sometime have smooth trajectory. Sometime struggle along the way. as show in this video 

https://youtu.be/ssxxNdnOZs8

Bagfile 
https://shorturl.at/eGZ36

And this is launch file with configuration

https://gist.github.com/fxprime/94c28b81890c24c8642b841b8b9136f9
https://gist.github.com/fxprime/bbae3381174efd4cdab870364eca852f

I don't sure it's machine limitation or some cause. 
Here is machine information

```
fx@ubuntu:~/jetsonUtilities$ python3 jetsonInfo.py 
NVIDIA NVIDIA Jetson Xavier NX Developer Kit
 L4T 35.3.1 [ JetPack UNKNOWN ]
   Ubuntu 20.04.6 LTS
   Kernel Version: 5.10.104-tegra
 CUDA 11.4.315
   CUDA Architecture: 7.2
 OpenCV version: 4.5.5
   OpenCV Cuda: YES
 CUDNN: 8.6.0.166
 TensorRT: 8.5.2.2
 Vision Works: NOT_INSTALLED
 VPI: 2.2.7
 Vulcan: 1.3.204
```

Can someone help me to improve the performance or something i missed?
Thank you advanced.

ripdk12, 2023-06-24:
This is a long shot, but are you running this from an SD card?
I personally experienced quite a performance boost after switching to SSD.

I also use a Xavier NX and I get much better performance, often 90hz VIO data.

tdnet12434, 2023-06-25:
> This is a long shot, but are you running this from an SD card? I personally experienced quite a performance boost after switching to SSD.
> 
> I also use a Xavier NX and I get much better performance, often 90hz VIO data.

OMG. 90hz is much more than I expected. Are you also running on docker? If so, it's impressive!
I also use SSD run overall Ubuntu (installed directly from sdkmanager to SSD)
Still can't figure out why performance is worse like this.

jaiveersinghNV, 2023-06-26:
Based on one of your launchfiles, it seems like you might be running two instances of Visual SLAM concurrently? If so, have you tried running just one instance and observing the performance there?

You might also install the [`jetson_stats` tool](https://github.com/rbonghi/jetson_stats) to monitor the system utilization during the operation of your launch graph. It could be the case that some non-Isaac ROS background process is consuming too much CPU or memory bandwidth.

closed 2023-07-26 by jaiveersinghNV

jenniferaustin326, 2024-05-02:
Wish I never traded here, they locked me out of my account and wouldn't let me withdraw the $200k I invested. It was really hard, I couldn't maintain my mortgage and had to sell my house. But with help from a Professional, I was able to get back my money. Reach out for help, if you're facing the same issues with them and need your money out. robertslee618 @ gmail.com... you can also reach out to her via (Whatsapp) +1 (217) 610 2766..