NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #98: Load and Localize

tom-deBach, 2023-06-14:
Hi,
I have some questions about the LoadMapAndLocalize action. 
I tried it several times and found it quite hard to correctly pinpoint the location with the prior guess of the pose, so the map would not load (The best result I got was circa 1.5m radius around the correct position).

Am I doing something wrong or do I really have to know my position this close to load a map? I would have expected, that it maybe takes more time if the guess is not too good, or takes more landmarks than the current into account.

I found this parameter, but it seems, this is still in development: 
[image]
[image]

This seems to include, what i would hope for. Are there current plans to implement this?

Kind regards and thanks in advance

jaiveersinghNV, 2023-06-20:
We're working on this feature, but we don't have any updates on when it will be delivered. Could you try using a feature in your task environment to align the start of your mapping and then re-align the robot there during the load map stage?

tom-deBach, 2023-06-21:
First of all: Thanks for your work :) I really like the packages and combination with VSLAM, Nvblox and nav2.

Unfortunately I don't have such a feature. Our goal is to load a map into a electronic travel aid for visually impaired people, so we cant guarantee a "starting position" as you would have with an AGV for example and as it would be needed to localize with the "load and localize" function, if I understood correctly. 

Could the isaac_ros_map_localization package be of use with a stereo camera, instead of a lidar?

And a small heads up: This does not really concern you at nvidia directly as it is more of a Intel topic, but we had problems with the auto exposure not working together with the laser_emitter_on_off function of the realsense splitter on a Realsense D435. 

Kind regards

jaiveersinghNV, 2023-06-26:
The `isaac_ros_map_localization` package is currently lidar-only, and it will not work with a stereo camera. We're working on speeding up the `LoadMapAndLocalize` action to make it feasible for you to run a grid search over multiple candidate initial guesses.

If you're experiencing issues with the RealSense camera, you may be able to find better support for it on the [Intel forums](https://support.intelrealsense.com/hc/en-us/community/topics). Note that the RealSense splitter we provide is explicitly designed to support simultaneous usage of VSLAM and Nvblox; if you're only using one of the two, you may be able to avoid the problem by removing the splitter from your graph entirely.