NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #104: Using camera Intel RealSense D435i as camera tracking

lycn2020, 2023-07-24:
Hello,
I am using on the last month jetson nano with Intel RealSense Tracking Camera T265 and CUAV v5+ on my quad drone for using it as Navigation when GPS signal not found or loss for some time using "vision_position_estimate" message that send it to Ardupilot via mavlink protocol and I found this error " T265: VISO jump detected" .
Now I want to use Jetson AGX Xavier with Intel RealSense D435i as camera tracking to work as Intel RealSense Tracking Camera T265 and to give me position estimate data.
How I can do it? any one help me .
Thank you for your help!

swapnesh-wani-nvidia, 2023-07-24:
Hello, in order to use your D435i for tracking, you could use Isaac ROS Visual Slam package. Take a look at this [tutorial](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/main/docs/tutorial-realsense.md) which shows how to use Intel RealSense camera.

lycn2020, 2023-07-24:
Thank you swapnesh-wani-nvidia I am using  Isaac ROS Visual Slam package and I am running my camera using "ros2 launch isaac_ros_visual_slam isaac_ros_visual_slam_realsense.launch.py" code but when I move camera up and down i found this message "[visual_slam_node]: Visual tracking is lost" , I don't know why? .
I want to read data position using python scripts .
Thank you for your time .