NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #107: Isaac SLAM with Luxonis OAK camera

martinlucan, 2023-08-09:
Hi there,
I intend to use Isaac Visual SLAM in conjunction with Luxonis OAK-D-Pro camera. As far as I know, there is no direct integration of this camera in Isaac package. Is there any general approach how to configure Isaac stack to work with any suitable stereo camera?

swapnesh-wani-nvidia, 2023-08-14:
Isaac ROS Visual Slam expects a pair of synchronized stereo images. Take a look and the following launch files on how to connect left and right image topics along with respective camerainfo topics 

1. [isaac_ros_visual_slam_realsense.launch.py](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/main/isaac_ros_visual_slam/launch/isaac_ros_visual_slam_realsense.launch.py)
2. [isaac_ros_visual_slam_hawk.launch.py](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/main/isaac_ros_visual_slam/launch/isaac_ros_visual_slam_hawk.launch.py)

AlexKlimaj, 2023-08-18:
I would like to get this running as well with the Oak D Pro Wide. It looks like this example might be all that's needed to get the stereo outputs.
https://github.com/luxonis/depthai/blob/main/depthai_sdk/examples/streaming/ros_publishing.py

bradenwagstaff, 2023-08-22:
So I've been playing with the ros_publishing.py script and am having a hard time getting it to mesh with Isaac VSLAM. If you look at the launch file for their [realsense example](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/main/isaac_ros_visual_slam/launch/isaac_ros_visual_slam_realsense.launch.py) it subscribes to the topics of the left and right camera, but also the camera_info for both. 

As far as I can tell, the depthai script doesn't publish camera info on a topic. All I can get out of it is
/color/compressed
/imu
/left/compressed
/right/compressed

I've been going through the Depthai SDK documentation and can't find a way to publish that information. Please prove me wrong, but I'm going to move to Realsense to use Isaac VSLAM

Serafadam, 2023-08-28:
Hi @bradenwagstaff  @AlexKlimaj Depthai SDK currently proves limited functionality in terms of ROS publishing, when it comes to the ROS integration we recommend using `depthai-ros` packages, either via Docker images or by building it from source (this may change this year when Jetpack will support 22.04). Currently there is no official support for Isaac SLAM, but that might change in near future (you can also refer to the community integration [here](https://github.com/msardonini/oakd_isaac_ros/tree/main))

closed 2023-10-30 by jaiveersinghNV