NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #108: vslam with custom camera

lgjonathan, 2023-08-10:
Hi, I have successfully ran the vslam quickstart with the realsense camera.
But there's a problem when I try to run the vslam with custom cameras.
I have tried with 2 Arducam OV9281 and published following topics.  They seemed to be published successfully when I echo them.
- `/stereo_camera/left/image`
- `/stereo_camera/left/camera_info`
- `/stereo_camera/right/image`
- `/stereo_camera/right/camera_info`

After publishing, I launched the `/visual_slam_node`. The  `/visual_slam_node` is subscribing to the abovementioned topics but the fixed frame is saying "`No tf data. Actual error: Frame [map] does not exist`". So my guess is that the vslam node is not calculating the stereo images and stuck somewhere in the process. Is there a way for me to debug where the problem is?

Same problem occurs when I use 2 webcams instead. I am wondering if I should do more in my end than just publishing the topics. I am quite new to the ROS, so any suggestions or directions will be appreciated.

lgjonathan, 2023-08-11:
I found out that I needed to publish the camera_info with the same rate of image.
After fixing that, It's working.

closed 2023-08-11 by lgjonathan

hemalshahNV, 2023-08-11:
Yes, camera images by ROS convention publish as sensor_msgs::Image and a matching CameraInfo on a separate topic. You could get away with one CameraInfo for the entire stream of sensor_msgs::Image, but it is also possible that the calibration parameters could change in CameraInfo dynamically say because you're currently calibrating the camera or something else.