NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #110: Build failure on Humble

sandeepdutta, 2023-08-18:
Hello All,

I am trying to build on Humble , and I get this error.
Starting >>> isaac_ros_visual_slam
--- stderr: isaac_ros_visual_slam                         
CMake Error at /home/sandeep/Desktop/ros2_humble/install/ament_cmake_core/share/ament_cmake_core/cmake/index/ament_index_get_resource.cmake:62 (message):
  ament_index_get_resource() called with not existing resource ('cuvslam'
  'isaac_ros_nitros')
Call Stack (most recent call first):
  CMakeLists.txt:37 (ament_index_get_resource)

Do I need to install some other packages.

Thank you
Sandeep

swapnesh-wani-nvidia, 2023-08-21:
Take a look at step 2 of [quickstart](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam#quickstart). You have to clone `isaac_ros_nitros` to bring in `cuvslam` library.

sandeepdutta, 2023-08-21:
Thank you so much Swapnesh, I will try this, Is there a way to run this natively i.e. without docker ?
Sandeep

mikolodz, 2023-10-10:
> Thank you so much Swapnesh, I will try this, Is there a way to run this natively i.e. without docker ? Sandeep

Hi, I've had the same issue as @sandeepdutta and after cloning the isaac_ros_nitros as @swapnesh-wani-nvidia suggested, I'm getting another error:

```
---
Finished <<< isaac_ros_nitros_interfaces [2.33s]
Starting >>> isaac_ros_nitros
--- stderr: isaac_ros_nitros                          
CMake Error at CMakeLists.txt:46 (find_package):
  By not providing "Findvpi.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "vpi", but
  CMake did not find one.

  Could not find a package configuration file provided by "vpi" with any of
  the following names:

    vpiConfig.cmake
    vpi-config.cmake

  Add the installation prefix of "vpi" to CMAKE_PREFIX_PATH or set "vpi_DIR"
  to a directory containing one of the above files.  If "vpi" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< isaac_ros_nitros [0.91s, exited with code 1]

Summary: 3 packages finished [3.85s]
  1 package failed: isaac_ros_nitros
  2 packages had stderr output: isaac_ros_nitros isaac_ros_nitros_interfaces
  15 packages not processed
```

I'm also trying to build it locally without luck (there are some other ROS2 projects on my workstation working properly).

Earlier, I tried to use your docker files as Quickstart suggests, but there are some problems e.g. cherry picking some commit from some fork in a not compatible way, using docker images which are already outdated and not present on docker hub etc. I've spent a day or two to figure it out, but still without luck to make the examples running.

That's why I decided to run it natively w/o docker.

By any chance, @swapnesh-wani-nvidia could you take a look at the docker files in your repo (used in the Quickstart) and validate if they are up-to-date? Maybe your team could fix them quickly - it's definitely something preventing people from getting into Isaac-Ros easily.

Thank you!

jaiveersinghNV, 2023-10-30:
Hi @mikolodz , we've recently updated our documentation and consolidated it on our new site [here](https://nvidia-isaac-ros.github.io/getting_started/dev_env_setup.html).

Could you please take a second look at our Docker setup instructions and see if it works for you this time? We provide a Docker image specifically because the web of dependencies required for Isaac ROS is very complicated; resolving them manually will be a challenging endeavor.

If you are sure that you want to run natively without Docker, then you should follow along each line of the `Dockerfile.x` file 'layers' and manually install the appropriate dependencies.

closed 2023-11-27 by jaiveersinghNV