NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #111: [component_container-1] [ERROR] !gtsam initial failed!

furkansariyildiz, 2023-08-23:
Hello everyone,

I am trying to use isaac_ros_visual_slam in docker image. I have Intel Realsense D435i Camera and I am running this "$ ros2 launch isaac_ros_visual_slam isaac_ros_visual_slam_realsense.launch.py" command. 

In 2-3 minutes, there is no error in SLAM but after this time, camera_link is jumping to random location and it returns "[component_container-1] [ERROR] !gtsam initial failed!" error. Thanks for everything!

jaiveersinghNV, 2023-10-23:
Could you please try again with the [latest release](https://nvidia-isaac-ros.github.io/releases/index#isaac-ros-2-0-0-october-18-2023)? This release should improve stability. if you're still experiencing issues, please provide a sample rosbag with data and we will try to reproduce the behavior on our end.

furkansariyildiz, 2023-11-06:
Hello @jaiveersinghNV, I will try again and write the results. Thanks.

closed 2023-11-27 by jaiveersinghNV

silence-moon, 2024-10-17:
I also get same error. I launch the file isaac_ros_visual_slam_realsense.launch.py with version of dp3.1 on ubuntu 20.04 system with cuda 11.4. Some logs are:
[component_container-1] [WARN] [1729127424.632447921] [visual_slam_node]: Visual tracking is lost
[component_container-1] [ERROR] !gtsam initial failed!
[component_container-1] [ERROR] !gtsam initial failed!
[component_container-1] [ERROR] !gtsam initial failed!
...
At the same time , camera_link is jumping to random location far away (0,0,0).
What do I need to do next？I am looking forward your apply.