NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #113: OAK-Pro VSLAM: Frame [map] does not exist

martinlucan, 2023-08-30:
Hi there,
I am trying to run Luxonis OAK-Pro camera in stereo mode with Isaac VSLAM, although the process stucks inside VSLAM node. The camera publishes non-rectified left/right images with corresponding camera_info topics and IMU data. According to rqt_graph, the visual_slam node subscribes to remapped topics, but outputs empty topics and map -> camera_link transform (Error: `Frame [map] does not exist`). Could you please check my launchfile and confirm the parameter configuration? Any opinions on the reason of missing transform?

I use following launchfile:
```
import launch
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    """Launch file to bring up visual slam node standalone."""
    visual_slam_node = ComposableNode(
        name='visual_slam_node',
        package='isaac_ros_visual_slam',
        plugin='isaac_ros::visual_slam::VisualSlamNode',
        remappings=[('stereo_camera/left/image', 'oak/left/image_raw'),
                    ('stereo_camera/right/image', 'oak/right/image_raw'),
                    ('stereo_camera/left/camera_info', 'oak/left/camera_info'),
                    ('stereo_camera/right/camera_info', 'oak/right/camera_info'),
                    ('visual_slam/imu', 'oak/imu/data')],
                                       
        parameters=[{
                    'use_sim_time': False,
                    'denoise_input_images': True,
                    'rectified_images': True,
                    'enable_slam_visualization': True,
                    'enable_observations_view': True,
                    'enable_landmarks_view': True,
                    'enable_imu_fusion': True, 
                    'enable_debug_mode': False,
                    'enable_verbosity': True,
                    'debug_dump_path': '/tmp/cuvslam',
                    'base_frame': 'oak-d-base-frame',
                    'map_frame': 'map',
                    'odom_frame': 'odom',
                    'input_imu_frame': 'oak_imu_frame'
                    }]
    )

    visual_slam_launch_container = ComposableNodeContainer(
        name='visual_slam_launch_container',
        namespace='',
        package='rclcpp_components',
        executable='component_container',
        composable_node_descriptions=[
            visual_slam_node
        ],
        output='screen'
    )

    return launch.LaunchDescription([visual_slam_launch_container])
```
rqt_graph output:

[image]

martinlucan, 2023-09-04:
@hemalshahNV @jaiveersinghNV @swapnesh-wani-nvidia guys, could you please take a look at this? Thank you in advance.

jaiveersinghNV, 2023-10-23:
Could you please try again on our [latest release](https://nvidia-isaac-ros.github.io/releases/index#isaac-ros-2-0-0-october-18-2023) and see if the problem is still happening?

If you're still seeing this issue, please provide a rosbag of recorded data from your camera that we can use to replicate the problem

Interpause, 2023-11-08:
I believe I found the error in the launch file.

https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/406bf6f82bf8bac5150ec5ce1e8e9186c2ae4308/isaac_ros_visual_slam/launch/isaac_ros_visual_slam_realsense.launch.py#L25-L27

Above creates the camera topics like `/camera/camera/infra1/image_rect_raw` and so on. However, below:

https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/406bf6f82bf8bac5150ec5ce1e8e9186c2ae4308/isaac_ros_visual_slam/launch/isaac_ros_visual_slam_realsense.launch.py#L69-L73

Remaps the subscribers to `/camera/infra1/image_rect_raw` and so on.

Seems this error is in most of the launch files. The simplest solution is to set `namespace=''`. This allows the visual slam node to become active and publish the tf.

swapnesh-wani-nvidia, 2023-11-20:
Hi, could you provide the full logs of your run. Looks like a broken transform tree issue. Could you verify the left camera and rigfht camera frame name? Also, when using the u-rectified images, please set this param to false
 `'rectified_images': False`