NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #114: SLAM node quits on bag play

JChunX, 2023-08-31:
Hi, I am trying to run the SLAM example on my machine (Ubuntu 22.04, CUDA 12.2, Driver 535.86.10)
The vslam launch works (`Loaded node '/visual_slam_node' in container '/visual_slam_launch_container'`) until I run the bag playback, upon which the process dies immediately. 

```
admin@dojo:/workspaces/isaac_ros-dev$ ros2 launch isaac_ros_visual_slam isaac_ros_visual_slam.launch.py
[INFO] [launch]: All log files can be found below /home/admin/.ros/log/2023-08-31-09-35-28-178280-dojo-4727
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [4738]
[component_container-1] [INFO] [1693488928.544081787] [visual_slam_launch_container]: Load Library: /workspaces/isaac_ros-dev/install/isaac_ros_visual_slam/lib/libvisual_slam_node.so
[component_container-1] [INFO] [1693488928.630549925] [visual_slam_launch_container]: Found class: rclcpp_components::NodeFactoryTemplate<isaac_ros::visual_slam::VisualSlamNode>
[component_container-1] [INFO] [1693488928.630590792] [visual_slam_launch_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<isaac_ros::visual_slam::VisualSlamNode>
[component_container-1] [INFO] [1693488928.644340992] [visual_slam_node]: cuVSLAM version: 11.0
[component_container-1] [INFO] [1693488928.648281972] [visual_slam_node]: Time taken by CUVSLAM_WarmUpGPU(): 0.003909
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/visual_slam_node' in container '/visual_slam_launch_container'
[component_container-1] [INFO] [1693488966.930735505] [visual_slam_node]: left_pose_right reading from CameraInfo
[component_container-1] [INFO] [1693488966.930773849] [visual_slam_node]: Baseline is : 0.119973
[component_container-1] [INFO] [1693488966.930795120] [visual_slam_node]: Use use_gpu: true
[component_container-1] [INFO] [1693488966.930802489] [visual_slam_node]: Enable IMU Fusion: false
[ERROR] [component_container-1]: process has died [pid 4738, exit code -6, cmd '/opt/ros/humble/install/lib/rclcpp_components/component_container --ros-args -r __node:=visual_slam_launch_container -r __ns:=/'].
```

It is also notable that the workspace had test failures:

```
Summary: 27 packages finished [3min 42s]
  13 packages had stderr output: isaac_ros_nitros_april_tag_detection_array_type isaac_ros_nitros_compressed_image_type isaac_ros_nitros_disparity_image_type isaac_ros_nitros_flat_scan_type isaac_ros_nitros_image_type isaac_ros_nitros_occupancy_grid_type isaac_ros_nitros_point_cloud_type isaac_ros_nitros_pose_array_type isaac_ros_nitros_pose_cov_stamped_type isaac_ros_nitros_tensor_list_type isaac_ros_occupancy_grid_localizer isaac_ros_pointcloud_utils isaac_ros_visual_slam
  13 packages had test failures: isaac_ros_nitros_april_tag_detection_array_type isaac_ros_nitros_compressed_image_type isaac_ros_nitros_disparity_image_type isaac_ros_nitros_flat_scan_type isaac_ros_nitros_image_type isaac_ros_nitros_occupancy_grid_type isaac_ros_nitros_point_cloud_type isaac_ros_nitros_pose_array_type isaac_ros_nitros_pose_cov_stamped_type isaac_ros_nitros_tensor_list_type isaac_ros_occupancy_grid_localizer isaac_ros_pointcloud_utils isaac_ros_visual_slam
```

But the stderr was unhelpful when I re-ran it with `colcon test --rerun-failed --output-on-failure`: 
```
--- stderr: isaac_ros_visual_slam                                       
Errors while running CTest
---
Finished <<< isaac_ros_visual_slam [14.6s]	[ with test failures ]
```

Any tips would be appreciated!

JChunX, 2023-08-31:
nvm - uncovered a driver issue upon machine restart. needed to reinstall drivers

closed 2023-08-31 by JChunX

rictsl, 2024-05-08:
Hi, I have the exact same issue right now, what drivers did you have to reinstall? My nvidia drivers seem to work, both `nvcc --version` and `nvidia-smi` return normal readings.