NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #117: In +WSL2 with Docker - ros2 launch isaac_ros_visual_slam isaac_ros_visual_slam.launch.py starts but nothing shows in RViz

Twenkid, 2023-09-18:


Twenkid, 2023-09-18:
I just built it again, building returns no errors, but testing returns an error from isaac_ros_visual_slam:

```bash
admin@docker-desktop:/workspaces/isaac_ros-dev$  colcon build --symlink-install
Starting >>> isaac_ros_common
Starting >>> isaac_ros_test
Finished <<< isaac_ros_common [1.23s]
Starting >>> isaac_ros_gxf
Starting >>> isaac_ros_nitros_interfaces
Starting >>> isaac_ros_apriltag_interfaces
Finished <<< isaac_ros_test [2.60s]
Starting >>> isaac_ros_pointcloud_interfaces
Finished <<< isaac_ros_gxf [1.68s]
Starting >>> isaac_ros_tensor_list_interfaces
Finished <<< isaac_ros_nitros_interfaces [4.43s]
Starting >>> isaac_ros_nitros
Finished <<< isaac_ros_pointcloud_interfaces [3.79s]
Starting >>> isaac_ros_visual_slam_interfaces
Finished <<< isaac_ros_apriltag_interfaces [5.22s]
Starting >>> isaac_ros_bi3d_interfaces
Finished <<< isaac_ros_nitros [2.77s]
Starting >>> isaac_ros_nitros_april_tag_detection_array_type
Finished <<< isaac_ros_tensor_list_interfaces [5.92s]
Starting >>> isaac_ros_nitros_camera_info_type
Finished <<< isaac_ros_bi3d_interfaces [3.74s]
Starting >>> isaac_ros_nitros_compressed_image_type
Finished <<< isaac_ros_nitros_april_tag_detection_array_type [1.90s]
Starting >>> isaac_ros_nitros_detection2_d_array_type
Finished <<< isaac_ros_nitros_camera_info_type [1.60s]
Starting >>> isaac_ros_nitros_disparity_image_type
Finished <<< isaac_ros_nitros_compressed_image_type [1.57s]
Starting >>> isaac_ros_nitros_flat_scan_type
Finished <<< isaac_ros_nitros_disparity_image_type [1.50s]
Starting >>> isaac_ros_nitros_image_type
Finished <<< isaac_ros_nitros_detection2_d_array_type [1.93s]
Starting >>> isaac_ros_nitros_imu_type
Finished <<< isaac_ros_nitros_image_type [1.58s]
Starting >>> isaac_ros_nitros_occupancy_grid_type
Finished <<< isaac_ros_nitros_flat_scan_type [1.76s]
Starting >>> isaac_ros_nitros_point_cloud_type
Finished <<< isaac_ros_nitros_imu_type [1.60s]
Starting >>> isaac_ros_nitros_pose_array_type
Finished <<< isaac_ros_visual_slam_interfaces [8.38s]
Starting >>> isaac_ros_nitros_pose_cov_stamped_type
Finished <<< isaac_ros_nitros_occupancy_grid_type [1.78s]
Starting >>> isaac_ros_nitros_std_msg_type
Finished <<< isaac_ros_nitros_point_cloud_type [1.79s]
Starting >>> isaac_ros_nitros_tensor_list_type
Finished <<< isaac_ros_nitros_pose_array_type [1.80s]
Starting >>> isaac_ros_visual_slam
Finished <<< isaac_ros_nitros_pose_cov_stamped_type [1.81s]
Finished <<< isaac_ros_nitros_std_msg_type [1.60s]
Finished <<< isaac_ros_nitros_tensor_list_type [1.58s]
Finished <<< isaac_ros_visual_slam [1.96s]

Summary: 25 packages finished [18.0s]
admin@docker-desktop:/workspaces/isaac_ros-dev$ ros2 launch isaac_ros_visual_slam isaac_ros_visual_slam.launch.py^C
admin@docker-desktop:/workspaces/isaac_ros-dev$   source install/setup.bash
admin@docker-desktop:/workspaces/isaac_ros-dev$
admin@docker-desktop:/workspaces/isaac_ros-dev$
admin@docker-desktop:/workspaces/isaac_ros-dev$ colcon test --executor sequential
Starting >>> isaac_ros_common
Finished <<< isaac_ros_common [2.38s]
Starting >>> isaac_ros_test
Finished <<< isaac_ros_test [14.9s]
Starting >>> isaac_ros_apriltag_interfaces
Finished <<< isaac_ros_apriltag_interfaces [1.14s]
Starting >>> isaac_ros_bi3d_interfaces
Finished <<< isaac_ros_bi3d_interfaces [1.14s]
Starting >>> isaac_ros_gxf
Finished <<< isaac_ros_gxf [1.43s]
Starting >>> isaac_ros_nitros_interfaces
Finished <<< isaac_ros_nitros_interfaces [0.90s]
Starting >>> isaac_ros_pointcloud_interfaces
Finished <<< isaac_ros_pointcloud_interfaces [0.90s]
Starting >>> isaac_ros_tensor_list_interfaces
Finished <<< isaac_ros_tensor_list_interfaces [1.17s]
Starting >>> isaac_ros_visual_slam_interfaces
Finished <<< isaac_ros_visual_slam_interfaces [1.13s]
Starting >>> isaac_ros_nitros
Finished <<< isaac_ros_nitros [8.78s]
Starting >>> isaac_ros_nitros_april_tag_detection_array_type
Finished <<< isaac_ros_nitros_april_tag_detection_array_type [8.12s]
Starting >>> isaac_ros_nitros_camera_info_type
Finished <<< isaac_ros_nitros_camera_info_type [8.01s]
Starting >>> isaac_ros_nitros_compressed_image_type
Finished <<< isaac_ros_nitros_compressed_image_type [8.00s]
Starting >>> isaac_ros_nitros_detection2_d_array_type
Finished <<< isaac_ros_nitros_detection2_d_array_type [8.22s]
Starting >>> isaac_ros_nitros_disparity_image_type
Finished <<< isaac_ros_nitros_disparity_image_type [8.03s]
Starting >>> isaac_ros_nitros_flat_scan_type
Finished <<< isaac_ros_nitros_flat_scan_type [8.14s]
Starting >>> isaac_ros_nitros_image_type
Finished <<< isaac_ros_nitros_image_type [7.09s]
Starting >>> isaac_ros_nitros_imu_type
Finished <<< isaac_ros_nitros_imu_type [6.93s]
Starting >>> isaac_ros_nitros_occupancy_grid_type
Finished <<< isaac_ros_nitros_occupancy_grid_type [8.00s]
Starting >>> isaac_ros_nitros_point_cloud_type
Finished <<< isaac_ros_nitros_point_cloud_type [22.9s]
Starting >>> isaac_ros_nitros_pose_array_type
Finished <<< isaac_ros_nitros_pose_array_type [6.96s]
Starting >>> isaac_ros_nitros_pose_cov_stamped_type
Finished <<< isaac_ros_nitros_pose_cov_stamped_type [8.03s]
Starting >>> isaac_ros_nitros_std_msg_type
Finished <<< isaac_ros_nitros_std_msg_type [7.91s]
Starting >>> isaac_ros_nitros_tensor_list_type
Finished <<< isaac_ros_nitros_tensor_list_type [9.10s]
Starting >>> isaac_ros_visual_slam
--- stderr: isaac_ros_visual_slam
Errors while running CTest
Output from these tests are in: /workspaces/isaac_ros-dev/build/isaac_ros_visual_slam/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
---
Finished <<< isaac_ros_visual_slam [21.4s]      [ with test failures ]

Summary: 25 packages finished [3min 1s]
  1 package had stderr output: isaac_ros_visual_slam
  1 package had test failures: isaac_ros_visual_slam
 ```

swapnesh-wani-nvidia, 2023-09-18:
Hello, thank you for reaching out.

For `--runtime nvidia` flag not working, could you confirm that [step 1 of _Quickstart_](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam#quickstart) section is done? Also, what is the version of docker that you are running on Ubuntu 22.04 (inside WSL)?

For the crash, are you observing it instantaneously or after a few seconds? Are you seeing any output on the `visual_slam/tracking/odometry` topic?

Twenkid, 2023-09-20:
Hi, thanks for the comment and the suggestions! 

I installed the repos there: ~/workspaces/isaac_ros-dev$

Also: 
```
echo $ISAAC_ROS_WS
```
points there.

```
 docker --version
Docker version 24.0.6, build ed223bc
```

Re docker, I read contradicting suggestions, some people advice to turn off the service in WSL as Windows is translating it from the Windows Desktop version. 

It is running now:
````
```
sudo systemctl status docker
status docker
...
● docker.service - Docker Application Container Engine
     Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-09-17 21:52:16 EEST; 2 days ago
TriggeredBy: ● docker.socket
       Docs: https://docs.docker.com
   Main PID: 12626 (dockerd)
      Tasks: 11
     Memory: 44.6M
     CGroup: /system.slice/docker.service
             └─12626 /usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock

Sep 17 21:52:15 DESKTOP-LQSDBBS dockerd[12626]: time="2023-09-17T21:52:15.640754187+03:00" level=info msg="Default brid>
Sep 17 21:52:15 DESKTOP-LQSDBBS dockerd[12626]: time="2023-09-17T21:52:15.782300389+03:00" level=info msg="Loading cont>
Sep 17 21:52:15 DESKTOP-LQSDBBS dockerd[12626]: time="2023-09-17T21:52:15.799770290+03:00" level=warning msg="WARNING: >
Sep 17 21:52:15 DESKTOP-LQSDBBS dockerd[12626]: time="2023-09-17T21:52:15.799819590+03:00" level=warning msg="WARNING: >
Sep 17 21:52:15 DESKTOP-LQSDBBS dockerd[12626]: time="2023-09-17T21:52:15.799827790+03:00" level=warning msg="WARNING: >
Sep 17 21:52:15 DESKTOP-LQSDBBS dockerd[12626]: time="2023-09-17T21:52:15.799833090+03:00" level=warning msg="WARNING: >
Sep 17 21:52:15 DESKTOP-LQSDBBS dockerd[12626]: time="2023-09-17T21:52:15.799850290+03:00" level=info msg="Docker daemo>
Sep 17 21:52:15 DESKTOP-LQSDBBS dockerd[12626]: time="2023-09-17T21:52:15.799885890+03:00" level=info msg="Daemon has c>
Sep 17 21:52:16 DESKTOP-LQSDBBS dockerd[12626]: time="2023-09-17T21:52:16.034339194+03:00" level=info msg="API listen o>
Sep 17 21:52:16 DESKTOP-LQSDBBS systemd[1]: Started Docker Application Container Engine.
```
````

*On Windows it's 4.3.2, Docker Engine 20.10.11*

[image]

Oh, I thought it's supposed to update itself automatically. 
I guess that could be an incompatibility issue?
Now I see I have to manually install a new version.

>For the crash, are you observing it instantaneously or after a few seconds? Are you seeing >any output on the visual_slam/tracking/odometry topic?

It was crashing after running the ros bag. I couldn't make RViz show anything. I rerun it a bit earlier but it crashed immediately after starting the rosbag replay, so I guess there's no time to send/capture a proper message.

I'll update it and try again, starting with the original docker command.

Twenkid, 2023-09-20:
BTW, I missed to mention the GPU: it's a modest 750 Ti, but it supports CUDA 12.2 and compute capabilities 5.0. The repo suggests: "x86_64 | NVIDIA GPU | Ubuntu 20.04+CUDA 11.8+".

Could this be a problem, is it supposed to run with this amount of GPU RAM?

```
 nvidia-smi
Wed Sep 20 16:40:57 2023
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.54.04              Driver Version: 536.23       CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce GTX 750 Ti      On  | 00000000:01:00.0  On |                  N/A |
| 33%    0C    P0               1W /  38W |    508MiB /  2048MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+

+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A        23      G   /Xwayland                                 N/A      |
|    0   N/A  N/A        24      G   /Xwayland                                 N/A      |
+---------------------------------------------------------------------------------------+
```

```
 nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2021 NVIDIA Corporation
Built on Thu_Nov_18_09:45:30_PST_2021
Cuda compilation tools, release 11.5, V11.5.119
Build cuda_11.5.r11.5/compiler.30672275_0
```

Twenkid, 2023-09-26:
I updated docker inside WSL to 24.0.6 and repeated the procedure, with the same results... I managed to run the container, RViz opened, the slam launched, but as soon as the rosbag was played, the slam node died (with the original docker start script "nvidia"). The tests of the building process again returns error in the package, all the rest passes.

The last thing I tried was setting the:
```
{
    "runtimes": {
        "nvidia": {
            "path": "/usr/bin/nvidia-container-runtime",
            "runtimeArgs": []
        }
    },
    "default-runtime": "nvidia"
}
```

Script in the Windows Docker

The default was:

```
{
  "builder": {
    "gc": {
      "defaultKeepStorage": "20GB",
      "enabled": true
    }
  },
  "experimental": false
}
```

Again an error in colcon test.

I guess I'll have to move to a real Linux.

Twenkid, 2023-09-26:
I managed to run the example on a proper native Linux with a powerful GPU (but I see in nvidia-smi that it uses just hundreds  of MBs), with one small issue: the image windows didn't show in RViz, I opened one  with rqt_image_view.

[image]

swapnesh-wani-nvidia, 2023-10-02:
Great that you are able to run with a newer GPU. For the image not shown, could you check if RViz is subscribing to the correct image topic? For WSL2 case, your GPU - GTX 750 Ti is based on older architecture and it is not compatible with the cuVSLAM library.

Twenkid, 2023-10-10:
Thanks for the clarification about the GPU requirements! Unfortunately I can't test it on that machine anymore, but I guess the problem was what you said, probably I've missed to add the image topic, if it wasn't included automatically.

Twenkid, 2023-10-10:
What are the minimum GPU requirement/architecture? I can't find explicit data so far. There's info about RTX for Isaac SIM, but this one uses RViz and ROS. My test was on an RTX 3090.

jaiveersinghNV, 2023-10-30:
You can find the required hardware for running Isaac ROS on x86 [here](https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/compute/index.html#x86-platforms).

Twenkid, 2023-11-02:
OK, so Volta+ and Geforce RTX. I managed to run the demo on a 3090. Thanks

closed 2023-11-27 by jaiveersinghNV

GateraGael, 2024-04-12:
Hello were you able to get this very basic package to work with your initial setup ([IsaacSim-ros_workspace](https://github.com/NVIDIA-Omniverse/IsaacSim-ros_workspaces.git)) ? Windows 11 (Isaac Sim), ROS2 (Docker on WSL2) ?
From what I am finding is that 1) ISAAC Sim has to run on the same computer as where ROS2 is running (even when you specify --network host for docker it doesn't communicate with the running isaac sim 2) WSL2 does not support Vulkan therefore this cannot work (https://forums.developer.nvidia.com/t/my-issac-sim-at-omniverse-in-wsl2-can-not-launch/246932).

Twenkid, 2024-04-18:
Hi, I was on Win10 and was following the instruction for the specific example, it was crashing as explained. It was ros2 (isaac is in ros2 only, right?), but I think not IsaacSim, just isaac_ros (if working, it should be with the ros2 visualisation: rviz2 etc.); it shouldn't have run either because the GPU on the Win machine was a GF 750 TI, but I haven't found a clear info about the  requirements.