NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #118: how do i calculate map->base_link

tuch77731, 2023-09-21:
I very like this work. It's so amazing. I have some problem.

I want to fuse some difference sensor to make map->base_link, because sometime visual is not reliable.
I find odometry, vo_pose_covariance and vo_pose, but they are base on odom frame.
Is there slam_pose_covariance or slam_pose base on map frame?

Second problem is that Landmarks map suddenly change locations very large when my car run long time, and my map->base_link change very large distence.

Final problem is that the frequency of TF of map->odom published by VSLAM  is not stable. The average is 0.02 seconds, but sometimes it can be 0.3~0.5 seconds. It cause nav2 not work smooth.

Thank you.

swapnesh-wani-nvidia, 2023-10-23:
Hello, could you clarify if you are trying to fuse the map->base_link poses from different sensors? Usually, you fuse odom->base_link poses from different sensors.
We do provide slam pose (map->base_link) as part of this hierarchy map->odom->base_link. We update map->odom when there is a slam pose output.

Also, can you attach some logs, screenshots, rosbag to try and reproduce the issue?

tuch77731, 2023-11-09:
I found the cause of the delay problem.
I use this command to solve it.
sudo sysctl -w net.core.rmem_max=104857600 net.core.rmem_default=104857600

closed 2023-11-09 by tuch77731