NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #119: bluetooth library not found - Orin Nano

leclair-7, 2023-09-23:
I'm following the quickstart section: `https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/main/README.md#quickstart` and I'm getting an error where it's cmake is not finding the bluetooth library when building the workspace. I verified bluetooth is on in the os settings, toggled bluetooth. The file it's failing on is `joystick_drivers/wiimote/CMakeLists.txt`.

```cmake
--- stderr: wiimote                                                                                                                                               
CMake Error at wiimote-extras.cmake:19 (message):
  bluetooth library not found.
Call Stack (most recent call first):
  CMakeLists.txt:18 (include)
```

I flashed a microSD card with : https://nvidia-ai-iot.github.io/jetson_isaac_ros_visual_slam_tutorial/sdcard.html and booted, verified realsense-viewer runs on a D455.

leclair-7, 2023-09-23:
Fixed the bluetooth issue; looks like the isaac_ros_dev-aarch64 base image should get these added 
```
sudo apt-get install -y libspnav-dev
sudo apt-get install -y libbluetooth-dev
sudo apt-get install -y libcwiid1 libcwiid-dev
```

swapnesh-wani-nvidia, 2023-10-02:
Thank you for the suggestion we will try to add these packages in the future release.