NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #120: Causes/Solutions for "Delta between current and previous frame [] is above threshold []"

madgrizzle, 2023-10-08:
I've just was able to get this running with a Realsense D435i and I'm finding I'm getting some warning messages, sometimes continuously:
```
Delta between current and previous frame [33.381376] is above threshold [22.000000]
```
What are the causes for this and are there any recommended solutions?  It seems worse when running RVIZ.

dakejahl, 2023-10-09:
I also would appreciate an explanation about this. I understand that `img_jitter_threshold_ms` is the parameter controlling these warnings. The default value is 22ms (45hz) and my realsense is configured for 90fps. It spams under load with rviz2 or nvblox but also occurs spuriously with nothing but vslam running. What are the implications of continuously missing frames? I am seeing deltas as high as 150ms which seems horrible considering the 90fps of the camera.

swapnesh-wani-nvidia, 2023-10-23:
Thank you for bringing this up. Usually, when we see this warning message, it is generally because of RViz running in parallel. What it means is that cuVSLAM library didn't get the pair of images delivered to it under the set threshold i.e. the delta between the previously delivered pair of frames and the current pair is more than the set threshold.

Could you confirm if RealSense is indeed running at the desired fps profile?
`ros2 topic hz --window 100 <image_topic_name>`

Also, could you try switching DDS to Cyclone. It might be the communication between RealSense node and Isaac ROS VSLAM node.

madgrizzle, 2023-10-23:
For me, I can confirm that when I tested it using:
`ros2 topic hz --window 100 <image_topic_name>`
that I achieved 90 fps (89.99 or something reported)

I cannot readily switch to cyclone, as much as I really like to, because of issues with micro-ros compatibility (unless someone knows how to use micro-ros with cyclone)

vislero, 2023-12-28:
I have both the Orin AGX developer kit and the ORIN-NX 16GB (Seeed Studio reComputer J40). When I tried with the latter, I encountered the same issue (no issue on the AGX).

Additionally, even when I turned off RViz and only executed 'ros2 topic hz /visual_slam/tracking/odometry,' I encountered the same problem."

ishakbhatt-h3d, 2024-11-11:
Running into the same problem here and it doesn't seem that there are other solutions around. Would appreciate any guidance!

ahirane, 2025-02-14:
Same here ...  running only visual slam on Orin NX and get this message frecuentrly ...