NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #123: 'get_all_poses' services is posting scaled timestamps

lgjonathan, 2023-10-19:
When I call `visual_slam/get_all_poses` services, I'm supposed to get back list of timestamped poses.
I found out that the timestamp of the poses are scaled by 1000.
The posted time stamp.sec, stamp.nsec are both scaled by 1000.
I'm suspecting that somewhere in the `GetAllPoses` source code has a wrong conversion between seconds and milliseconds.

swapnesh-wani-nvidia, 2023-10-23:
Hello, could you send me the logs and the expected timestamps of the poses that you should be getting? What was the `max_count` value when you called the service?

lgjonathan, 2023-10-24:
What I did was:
1. record RealSense Node (images, camera_info) into ROSBAG.
2. spin VSLAM node
3. play the ROSBAG
4. call get_all_poses service

Below are the logs.
```
requester: making request: isaac_ros_visual_slam_interfaces.srv.GetAllPoses_Request(max_count=2500)

(showing first three)

response:
isaac_ros_visual_slam_interfaces.srv.GetAllPoses_Response(success=True, 
poses=[geometry_msgs.msg.PoseStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=1697772, nanosec=225198289), frame_id=''), pose=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=-3.8812562053180955e-09, y=-5.446525896957155e-09, z=-8.265562123632719e-10), orientation=geometry_msgs.msg.Quaternion(x=0.0, y=0.0, z=0.0, w=1.0))), 
geometry_msgs.msg.PoseStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=1697772, nanosec=225220645), frame_id=''), pose=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=4.418022854224546e-06, y=1.6736096313252347e-06, z=3.3916685424628668e-06), orientation=geometry_msgs.msg.Quaternion(x=-4.2152878479843255e-07, y=4.585047469218542e-07, z=-2.199920930271903e-06, w=0.9999999925494194))), 
geometry_msgs.msg.PoseStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=1697772, nanosec=225231822), frame_id=''), pose=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=8.012561920622829e-07, y=-2.598832679723273e-06, z=7.860900950618088e-06), orientation=geometry_msgs.msg.Quaternion(x=-8.024351245694561e-07, y=2.521119768061908e-06, z=-1.9565751472327975e-06, w=1.0))),

......

(showing last three)

geometry_msgs.msg.PoseStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=1697772, nanosec=252984882), frame_id=''), pose=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=0.06112809106707573, y=0.086497001349926, z=0.1103435829281807), orientation=geometry_msgs.msg.Quaternion(x=0.006695529637307941, y=-0.0036027766201613158, z=-0.0022156175509504866, w=0.9999686400145432))), 
geometry_msgs.msg.PoseStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=1697772, nanosec=252996008), frame_id=''), pose=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=0.06112809106707573, y=0.086497001349926, z=0.1103435829281807), orientation=geometry_msgs.msg.Quaternion(x=0.006695516315656437, y=-0.00360278194858956, z=-0.0022156228897145195, w=0.9999686474653574))), 
geometry_msgs.msg.PoseStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=1697772, nanosec=253007135), frame_id=''), pose=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=0.06112809106707573, y=0.086497001349926, z=0.1103435829281807), orientation=geometry_msgs.msg.Quaternion(x=0.006695528705956159, y=-0.0036027730111731614, z=-0.002215646539274695, w=0.9999686400145432)))])
```

As you can see, I've requested 2500 poses and yet the "sec"' remains the same for all of them.
To get the reasonable timestamps, I had to do 
```
sec * 1000 + nanosec * (10**-6)
```

which indicates to me that the timestamps are offset by a order of 3.

lgjonathan, 2023-10-24:
below is the ROSBAG info
```
Files:             camera_front_0.db3
Bag size:          1.2 GiB
Storage id:        sqlite3
Duration:          31.518s
Start:             Oct 20 2023 11:23:43.232 (1697772223.232)
End:               Oct 20 2023 11:24:14.750 (1697772254.750)
Messages:          11146
Topic information: Topic: /camera/infra2/image_rect_raw | Type: sensor_msgs/msg/Image | Count: 2786 | Serialization Format: cdr
                   Topic: /camera/infra1/image_rect_raw | Type: sensor_msgs/msg/Image | Count: 2786 | Serialization Format: cdr
                   Topic: /camera/infra2/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 2786 | Serialization Format: cdr
                   Topic: /camera/infra1/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 2786 | Serialization Format: cdr
                   Topic: /tf_static | Type: tf2_msgs/msg/TFMessage | Count: 2 | Serialization Format: cdr
```

FrontlineLucian, 2024-05-27:
i just ran into the same issue and can confirm this