NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #125: stderr: nvblox    FindDoxygen not find

UserName-wang, 2023-10-20:
in a docker, which build by command:
cd ${ISAAC_ROS_WS}/src/isaac_ros_common && \
  ./scripts/run_dev.sh ${ISAAC_ROS_WS}

error after command in the docker:

cd /workspaces/isaac_ros-dev && \
  colcon build --symlink-install && \
  source install/setup.bash

admin@ubuntu:/workspaces/isaac_ros-dev$ cd /workspaces/isaac_ros-dev &&   colcon build --symlink-install &&   source install/setup.bash
Starting >>> isaac_ros_common
Starting >>> isaac_ros_test
Starting >>> nvblox_msgs                                                            
Starting >>> nvblox_ros_common                                                                           
Starting >>> nvblox                                                                                                                 
Starting >>> nvblox_cpu_gpu_tools
Starting >>> nvblox_performance_measurement_msgs                                                                                                                                  
Starting >>> network_performance_measurement                                                                                                                                          
Starting >>> nvblox_examples_bringup                                                                                                                                                  
Starting >>> nvblox_isaac_sim
Starting >>> odometry_flattener                                                                                                                                                        
--- stderr: nvblox                                                                                                                                                                           
CMake Warning at /usr/share/cmake-3.22/Modules/FindDoxygen.cmake:460 (message):
  Unable to determine doxygen version: No such file or directory
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindDoxygen.cmake:623 (_Doxygen_find_doxygen)
  /workspaces/isaac_ros-dev/build/nvblox/_deps/ext_stdgpu-src/doc/CMakeLists.txt:7 (find_package)

CMake Error at /usr/share/cmake-3.22/Modules/FindDoxygen.cmake:702 (message):
  Unable to generate Doxyfile template: No such file or directory
Call Stack (most recent call first):
  /workspaces/isaac_ros-dev/build/nvblox/_deps/ext_stdgpu-src/doc/CMakeLists.txt:7 (find_package)

make: *** [Makefile:2433: cmake_check_build_system] Error 1
---
Failed   <<< nvblox [2.77s, exited with code 2]
Aborted  <<< isaac_ros_test [3.05s]
Aborted  <<< nvblox_cpu_gpu_tools [2.71s]
Aborted  <<< nvblox_isaac_sim [2.42s]
Aborted  <<< nvblox_examples_bringup [2.48s]
Aborted  <<< isaac_ros_common [3.12s]
Aborted  <<< network_performance_measurement [3.77s]                                                                                                                   
Aborted  <<< nvblox_msgs [5.59s]                                                                                                                                              
Aborted  <<< odometry_flattener [6.76s]                                                                                                                                     
Aborted  <<< nvblox_performance_measurement_msgs [8.53s]                                                                                  
Aborted  <<< nvblox_ros_common [8.96s]

hemalshahNV, 2023-10-23:
Let me take a look. This may have gotten lost in the noise. Easy to add doxygen to the Isaac ROS Dev base image though if that's all this is.

UserName-wang, 2023-10-24:
Dear @hemalshahNV ,
Thank you for your reply!
Which version of  doxygen we need?

UserName-wang, 2023-11-09:
just for your reference, i solved this goxygen issue by:
sudo apt-get install flex 
sudo apt-get install bison 

git clone -b Release_1_9_1 https://github.com/doxygen/doxygen.git

cd doxygen
mkdir build
cd build
cmake -G "Unix Makefiles" ..
make -j10
sudo make install

and changed the file: /workspaces/isaac_ros-dev/build/nvblox/_deps/ext_stdgpu-src/doc/CMakeLists.txt:7 
find_package(Doxygen 1.9.1) to find_package(doxygen 1.9.1)

but there are some other error pup out:
Finished <<< nvblox_nav2 [46.4s]                                                                                                                                                             
Finished <<< nvblox_image_padding [37.0s]                                                                                                                                                                
--- stderr: nvblox_rviz_plugin                                                                                                                                                                              
make[2]: *** No rule to make target '/opt/ros/humble/install/opt/rviz_ogre_vendor/lib/libOgreMain.so', needed by 'libnvblox_rviz_plugin.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:138: CMakeFiles/nvblox_rviz_plugin.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< nvblox_rviz_plugin [41.0s, exited with code 2]
Aborted  <<< isaac_ros_visual_slam_interfaces [1min 5s]                                                                                                                         
Aborted  <<< isaac_ros_nitros [2min 25s]                                                                                                                       
Aborted  <<< doxygen [3min 53s]                                                                                       
Aborted  <<< nvblox [4min 19s]

UserName-wang, 2023-11-09:
As I cannot find which file specified the path of libOgreMain.so', I have to use a stupid command to copy all the files to:
sudo cp -r /opt/ros/humble/opt/rviz_ogre_vendor/lib/ /opt/ros/humble/install/opt/rviz_ogre_vendor/

And I made a little bit progress, until a new another error came out: 

/workspaces/isaac_ros-dev/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp: In member function ‘void nvblox::NvbloxNode::saveMap(std::shared_ptr<nvblox_msgs::srv::FilePath_Request_<std::allocator<void> > >, std::shared_ptr<nvblox_msgs::srv::FilePath_Response_<std::allocator<void> > >)’:
/workspaces/isaac_ros-dev/src/isaac_ros_nvblox/nvblox_ros/src/lib/nvblox_node.cpp:1047:39: error: ‘using element_type = class nvblox::Mapper’ {aka ‘class nvblox::Mapper’} has no member named ‘saveLayerCake’
 1047 |   response->success = static_mapper_->saveLayerCake(filename);
      |                                       ^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/nvblox_ros_lib.dir/build.make:202: CMakeFiles/nvblox_ros_lib.dir/src/lib/nvblox_human_node.cpp.o] Error 1
make[2]: *** [CMakeFiles/nvblox_ros_lib.dir/build.make:188: CMakeFiles/nvblox_ros_lib.dir/src/lib/nvblox_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:141: CMakeFiles/nvblox_ros_lib.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< nvblox_ros [56.4s, exited with code 2]
Aborted  <<< isaac_ros_nitros_correlated_timestamp_type [1min 17s]                                                      
Aborted  <<< isaac_ros_nitros_pose_array_type [1min 21s]                                                                   
Aborted  <<< isaac_ros_nitros_occupancy_grid_type [1min 22s]                                                               
Aborted  <<< isaac_ros_managed_nitros [1min 24s]                                                                           
Aborted  <<< isaac_ros_nitros_detection2_d_array_type [1min 28s]                                                           
Aborted  <<< isaac_ros_nitros_encoder_ticks_type [1min 24s]                                                               
Aborted  <<< isaac_ros_nitros_disparity_image_type [1min 25s]                                                             
Aborted  <<< isaac_ros_nitros_imu_type [1min 25s]                                                                         
Aborted  <<< isaac_ros_nitros_odometry_type [1min 28s]                                                                    
Aborted  <<< isaac_ros_nitros_detection3_d_array_type [1min 39s]                                                          
Aborted  <<< isaac_ros_pointcloud_utils [2min 5s]                                           

Summary: 35 packages finished [2min 30s]
  1 package failed: nvblox_ros
  11 packages aborted: isaac_ros_managed_nitros isaac_ros_nitros_correlated_timestamp_type isaac_ros_nitros_detection2_d_array_type isaac_ros_nitros_detection3_d_array_type isaac_ros_nitros_disparity_image_type isaac_ros_nitros_encoder_ticks_type isaac_ros_nitros_imu_type isaac_ros_nitros_occupancy_grid_type isaac_ros_nitros_odometry_type isaac_ros_nitros_pose_array_type isaac_ros_pointcloud_utils
  1 package had stderr output: nvblox_ros
  8 packages not processed

jaiveersinghNV, 2023-11-20:
Hi @UserName-wang ,

Could you confirm which version of Isaac ROS Nvblox you're using? We recently released [Isaac ROS 2.1](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_nvblox/releases/tag/v2.1.0).

I've gone through the [quickstart](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_nvblox/isaac_ros_nvblox/index.html#quickstart) again on a Jetson AGX Orin with JetPack 5.1.2 and Isaac ROS 2.1, and I was not able to reproduce the missing Doxygen issue you've reported here.

UserName-wang, 2023-11-21:
@jaiveersinghNV , Thank you for your reply!
I always try to use the latest version via git pull. 
As I said before, each time I tried, I have new errors, Now I got this error:
...
Get:27 http://ports.ubuntu.com/ubuntu-ports focal-security/main arm64 Packages [2,408 kB]                                                                                                                                                
Get:28 http://ports.ubuntu.com/ubuntu-ports focal-security/restricted arm64 Packages [49.0 kB]                                                                                                                                           
Get:29 http://ports.ubuntu.com/ubuntu-ports focal-security/universe arm64 Packages [1,022 kB]                                                                                                                                            
Get:16 http://packages.ros.org/ros2-testing/ubuntu focal/main arm64 Packages [1,172 kB]                                                                                                                                                  
Get:16 http://packages.ros.org/ros2-testing/ubuntu focal/main arm64 Packages [1,172 kB]                                                                                                                                                  
Get:16 http://packages.ros.org/ros2-testing/ubuntu focal/main arm64 Packages [1,172 kB]                                                                                                                                                  
Get:16 http://packages.ros.org/ros2-testing/ubuntu focal/main arm64 Packages [1,172 kB]                                                                                                                                                  
Fetched 22.5 MB in 7min 50s (47.8 kB/s)                                                                                                                                                                                                  
Reading package lists... Done
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit file:///etc/ros/rosdep/sources.list.d/nvidia-isaac.yaml
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml]:
	<urlopen error [Errno 111] Connection refused> (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml]:
	<urlopen error [Errno 111] Connection refused> (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml]:
	<urlopen error [Errno 111] Connection refused> (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml]:
	<urlopen error [Errno 111] Connection refused> (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml]:
	Failed to download target platform data for gbpdistro:
	<urlopen error [Errno 111] Connection refused>
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
ERROR: error loading sources list:
	<urlopen error <urlopen error [Errno 111] Connection refused> (https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml)>
 * Stopping hotplug events dispatcher systemd-udevd                                                                                                                                                                                [ OK ] 
 * Starting hotplug events dispatcher systemd-udevd                                                                                                                                                                                [ OK ] 
admin@ubuntu:/workspaces/isaac_ros-dev$ sudo apt-get install -y ros-humble-isaac-ros-visual-slam
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-humble-isaac-ros-visual-slam

...

funny!

jaiveersinghNV, 2023-11-27:
Could you provide some information about the network environment this machine is operating in?

I notice that the download speed for fetching the sources list was very slow. 
> Fetched 22.5 MB in 7min 50s (47.8 kB/s)

Are you operating within some restricted environment that might be limiting your ability to access outside resources? Could you try again while connected to a different network?

Given the fact that the sources that are failing are all ones maintained by ROS, and not by NVIDIA, I'm inclined to think that the issue here is not something specific to NVIDIA Isaac ROS. Could you also try installing a different package, perhaps `rolldice`:
```
sudo apt-get install rolldice
```

zhangzscn, 2023-12-25:
@jaiveersinghNV I have the same problem, Unable to install ros-humble-isaac-ros-visual-slam,But rolldice can be installed successfully.
error:
E: Unable to locate package ros-humble-isaac-ros-visual-slam