NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #129: Delta between current and previous frame, although left/right topics Hz is 40.

michaelnguyen11, 2023-11-03:
Hi authors,

I'm using the Luxonis OAK camera with **Jetpack 5.1.2** and **Isaac ROS 2.0.0** on Orin Nano Developer Kit to run Isaac Visual SLAM.

The setup is working well, I can get the position of the camera through RVIZ. However, I get some warning from `visual_slam_node` :
`[visual_slam_node]: Delta between current and previous frame [49.911854] is above threshold [30.000000]`, although the left and right rectified topics hz are 40 and the imu data topics hz is approx. 400.

[image]

I tested the Visual SLAM by moving the camera in a circle and put back to station, the output looks quite drift, current position is pretty far from initial position.
[image]

I checked [Luxonis source code](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/src/dai_nodes/stereo.cpp#L275), look like they verified whether left and right rectified frames are synchronized or not, before publishing the image_rect left/right topics.

I think the issue relate to Delta between current and previous frame and probably the queues for left/right are full so it leads to mismatch between current and previous frame.

Could the author guide me on how to avoid the `Delta between current and previous frame` issue ? Many thanks in advance !.

Best regards,
Hiep Nguyen

swapnesh-wani-nvidia, 2023-11-20:
Hello, could you share your launch file and full logs? The warnings for the time delta specifies that the current pair of frames arrived late. This may be because of the exact time filter drops messages. Also, check the QoS setting of the camera.

usmanNoor5, 2024-11-14:
Having the same issue