NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #130: Error when running Rviz inside the Nvidia docker container

Shivam7Sharma, 2023-11-03:
Hi, 
I installed the Nvidia runtime container and the Docker container properly following all the instructions but I get the following output when I run Rviz:

>   rviz2 -d src/isaac_ros_visual_slam/isaac_ros_visual_slam/rviz/default.cfg.rviz
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-admin'
MESA: warning: Driver does not support the 0xa720 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
MESA: warning: Driver does not support the 0xa720 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
MESA: warning: Driver does not support the 0xa720 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
MESA: warning: Driver does not support the 0xa720 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
[INFO] [1699038236.153340747] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1699038236.153407313] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[INFO] [1699038236.201446488] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1699038236.425814995] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1699038236.522004214] [rviz2]: Stereo is NOT SUPPORTED

When I run a rosbag file for slam it stops recording data after 5 seconds but rviz stays open.

Can anyone help me with this?

jaiveersinghNV, 2023-11-06:
Was `nvidia-smi` run inside the container, or is that just the output from your host machine? If you didn't already, could you please run `nvidia-smi` inside the container spawned with `run_dev.sh`, and provide that output? We want to be certain that the container is properly able to access your GPU.

Other things to check:
- Does `rviz2` by itself (ie, without using the prepared config file) launch properly?
- Are you able to run the quickstart of any other repo (`isaac_ros_apriltag` is a lightweight one to test)
- Do the automated tests pass when you run `colcon test --executor sequential` as described in [Quickstart Step 4](https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_visual_slam/isaac_ros_visual_slam/index.html#quickstart)?

Shivam7Sharma, 2023-11-11:
> Was `nvidia-smi` run inside the container, or is that just the output from your host machine? If you didn't already, could you please run `nvidia-smi` inside the container spawned with `run_dev.sh`, [...]
Nvidia-smi inside the container:
[image]

```
colcon test --executor sequential 
Starting >>> isaac_ros_common
Finished <<< isaac_ros_common [2.07s]           
Starting >>> isaac_ros_test
Finished <<< isaac_ros_test [15.4s]             
Starting >>> isaac_ros_apriltag_interfaces
Finished <<< isaac_ros_apriltag_interfaces [0.88s]            
Starting >>> isaac_ros_bi3d_interfaces
Finished <<< isaac_ros_bi3d_interfaces [0.85s]            
Starting >>> isaac_ros_gxf
Finished <<< isaac_ros_gxf [1.28s]            
Starting >>> isaac_ros_nitros_bridge_interfaces
Finished <<< isaac_ros_nitros_bridge_interfaces [0.85s]            
Starting >>> isaac_ros_nitros_interfaces
Finished <<< isaac_ros_nitros_interfaces [0.63s]            
Starting >>> isaac_ros_nova_interfaces
Finished <<< isaac_ros_nova_interfaces [0.86s]            
Starting >>> isaac_ros_pointcloud_interfaces
Finished <<< isaac_ros_pointcloud_interfaces [0.60s]            
Starting >>> isaac_ros_tensor_list_interfaces
Finished <<< isaac_ros_tensor_list_interfaces [0.86s]            
Starting >>> isaac_ros_visual_slam_interfaces
Finished <<< isaac_ros_visual_slam_interfaces [0.87s]             
Starting >>> isaac_ros_nitros
Finished <<< isaac_ros_nitros [7.36s]             
Starting >>> isaac_ros_nitros_april_tag_detection_array_type
Finished <<< isaac_ros_nitros_april_tag_detection_array_type [6.39s]             
Starting >>> isaac_ros_nitros_battery_state_type
Finished <<< isaac_ros_nitros_battery_state_type [6.16s]             
Starting >>> isaac_ros_nitros_camera_info_type
Finished <<< isaac_ros_nitros_camera_info_type [6.14s]             
Starting >>> isaac_ros_nitros_compressed_image_type
Finished <<< isaac_ros_nitros_compressed_image_type [6.23s]             
Starting >>> isaac_ros_nitros_correlated_timestamp_type
Finished <<< isaac_ros_nitros_correlated_timestamp_type [6.11s]                 
Starting >>> isaac_ros_nitros_detection2_d_array_type
Finished <<< isaac_ros_nitros_detection2_d_array_type [6.18s]                  
Starting >>> isaac_ros_nitros_detection3_d_array_type
Finished <<< isaac_ros_nitros_detection3_d_array_type [6.18s]                  
Starting >>> isaac_ros_nitros_disparity_image_type
Finished <<< isaac_ros_nitros_disparity_image_type [8.31s]                  
Starting >>> isaac_ros_nitros_encoder_ticks_type
Finished <<< isaac_ros_nitros_encoder_ticks_type [6.31s]                  
Starting >>> isaac_ros_nitros_flat_scan_type
Finished <<< isaac_ros_nitros_flat_scan_type [6.33s]                  
Starting >>> isaac_ros_nitros_image_type
Finished <<< isaac_ros_nitros_image_type [6.50s]                  
Starting >>> isaac_ros_nitros_imu_type
Finished <<< isaac_ros_nitros_imu_type [6.24s]                  
Starting >>> isaac_ros_nitros_occupancy_grid_type
Finished <<< isaac_ros_nitros_occupancy_grid_type [7.31s]                  
Starting >>> isaac_ros_nitros_odometry_type
Finished <<< isaac_ros_nitros_odometry_type [6.40s]                  
Starting >>> isaac_ros_nitros_point_cloud_type
Finished <<< isaac_ros_nitros_point_cloud_type [6.30s]                  
Starting >>> isaac_ros_nitros_pose_array_type
Finished <<< isaac_ros_nitros_pose_array_type [6.25s]                  
Starting >>> isaac_ros_nitros_pose_cov_stamped_type
Finished <<< isaac_ros_nitros_pose_cov_stamped_type [6.31s]                  
Starting >>> isaac_ros_nitros_std_msg_type
Finished <<< isaac_ros_nitros_std_msg_type [6.08s]                  
Starting >>> isaac_ros_nitros_tensor_list_type
Finished <<< isaac_ros_nitros_tensor_list_type [8.40s]                  
Starting >>> isaac_ros_nitros_twist_type
Finished <<< isaac_ros_nitros_twist_type [6.23s]                  
Starting >>> isaac_ros_visual_slam
Finished <<< isaac_ros_visual_slam [42.2s]                   
Starting >>> isaac_ros_managed_nitros
Finished <<< isaac_ros_managed_nitros [1.34s]                  
Starting >>> custom_nitros_image
Finished <<< custom_nitros_image [15.7s]                   
Starting >>> custom_nitros_string
Finished <<< custom_nitros_string [15.4s]                   

Summary: 36 packages finished [3min 59s]

```
Rviz opens normally with the following command
```
admin@shivam-ROG-Zephyrus-M16-GU604VI-GU604VI:/workspaces/isaac_ros-dev$ rviz2
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-admin'
MESA: warning: Driver does not support the 0xa720 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
MESA: warning: Driver does not support the 0xa720 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
MESA: warning: Driver does not support the 0xa720 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
MESA: warning: Driver does not support the 0xa720 PCI ID.
libGL error: failed to create dri screen
libGL error: failed to load driver: iris
[INFO] [1699732109.183035563] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1699732109.183101652] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[INFO] [1699732109.241789931] [rviz2]: Stereo is NOT SUPPORTED

```

I haven't tried any other repo such as isaac_ros_apriltag. I can do that if it will help with this issue.

jaiveersinghNV, 2023-11-20:
Could you confirm the specifications of your host computer? What operating system and version are you using? Are you using WSL?

Running `isaac_ros_apriltag`'s quickstart would be a good debugging step. Please try completing that quickstart and see if you're able to launch the AprilTag graph without any problems.

Shivam7Sharma, 2024-02-25:
Is it possible that the ROS bag given in the tutorial is only 5 seconds? Because I can see the map in Rviz but it is there for only like 5 seconds. Let me know.

jaiveersinghNV, 2024-03-06:
Yes, we ship a short rosbag to avoid consuming too much storage. Unlike our other tutorials, we can't use the `--loop` playback mode for Visual SLAM, because the "jump" from the end to the start of the data sequence will lead to degraded performance with our algorithm.

If you need a longer dataset for some other tests, you can always use our [R2B Dataset 2023](https://catalog.ngc.nvidia.com/orgs/nvidia/teams/isaac/resources/r2bdataset2023) sequences.

Shivam7Sharma, 2024-05-28:
Closing this because the issue is resolved

closed 2024-05-28 by Shivam7Sharma

Shivam7Sharma, 2024-06-12:
@doctorcolossus  I believe this wasn't an issue because the ROS bag file was very small.

Shivam7Sharma, 2024-06-12:
@doctorcolossus I think you can safely ignore those errors.