NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #131: VSLAM Map & SLAM Tool box box

sandeepdutta, 2023-11-06:
I have the visual SLAM working with a ZED camera, it can localizes very well.
Is there a way to match the coordinates of the VSLAM generated map to the map generated by
SLAM toolbox ? I would like to use the map->odom transform generated by VSLAM
and use the rest of the NAV2 stack ..

Thank you
Sandeep

hemalshahNV, 2023-11-06:
You can use VSLAM as an odometry source with Nav2 (odom->base_link), but using the vslam poses (which can be used to update map->odom) can be a bit problematic. The VSLAM map origin needs to be registered with you obstacle map (occupancy grid map created by SLAM toolbox) (i.e., need to find the transform between vslam_map and og_map). This can be done by adding in physical features to align with in the OGM (like two boards covered in Apriltags at right angles in a V that you can see in both the OGM and the visual frames) or by some other method.

robotx10, 2023-11-20:
@hemalshahNV Hi Hemal, when can we expect the release of the document? 
@sandeepdutta were you able to implement the 3D to 2D tf2 projection as mentioned by hemal? Is there any other efficient solution that we adapt to?

sandeepdutta, 2023-11-20:
Yes I was able to create a transform between the maps, it wasn't very stable. We are planning to use two camera's.
Currently it looks like the package  doesn't support two cameras . You can only merge the Visual odometry using EKF.
Sandeep

robotx10, 2023-11-20:
@sandeepdutta may I know whats the reason behind fusing two cameras (Maybe you might different use case). Since the visual Inertial odometry from isaac_ros_vlsam work pretty well for me. I am using odom->base_link for nav2. I am having concerns about map->odom link. I was using slam_toolbox localization mode for map->odom (which i would like to skip and go with map->odom from vslam)

have you checked this? for fusing both cameras?
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam/blob/main/isaac_ros_visual_slam/params/ekf_dual_nvslam.yaml