NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #132: Docker Error - Unknown or Invalid Runtime Name: Nvidia

Buddies-as-you-know, 2023-11-15:
I am encountering a runtime error with Docker when trying to use the Nvidia runtime. This issue arises despite having a successful output with an initial Docker command and making subsequent edits to the Docker configuration.

#### Steps to Reproduce
1. Run the following Docker command which executes successfully:
   ```bash
   sudo docker run --rm --runtime=nvidia --gpus all ubuntu nvidia-smi
   ```

2. Edit `/etc/docker/daemon.json` as follows:
   ```json
   {
       "runtimes": {
           "nvidia": {
               "path": "nvidia-container-runtime",
               "runtimeArgs": []
           }
       },
       "default-runtime": "nvidia"
   }
   ```

3. After making these changes, attempt to execute a script with the command:
   ```bash
   scripts/run_dev.sh ~/workspaces/isaac_ros-dev/
   ```

   This results in the following error:
   ```
   docker: Error response from daemon: unknown or invalid runtime name: nvidia.
   ```

#### Expected Behavior
The Docker container should recognize the Nvidia runtime without errors, especially since the initial command runs without issues.

#### Actual Behavior
The system throws an error stating "unknown or invalid runtime name: nvidia" when trying to run a script that utilizes Docker with the Nvidia runtime.

#### Environment
- Docker version: [specify your Docker version]
- Operating System: [specify your OS]
- Any other relevant environmental details

#### Attempts to Resolve
- Verified that the initial Docker command runs successfully.
- Checked the syntax and paths in the `daemon.json` file.
- Searched for similar issues in forums and GitHub Issues.

#### Request for Help
Could anyone provide insights or suggest potential solutions to resolve this runtime error? Any advice or guidance would be greatly appreciated.

hemalshahNV, 2023-11-17:
It looks like you may not have `nvidia-container-toolkit` installed. See [here](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html#installing-with-apt) for instructions on how to install on your x86_64 system running Jammy.

Buddies-as-you-know, 2023-11-17:
We installed nvidia-container-toolkit.

solix, 2023-11-17:
I am experiencing same issue, nvidia-container-toolkit is also installed.

weirdsim14, 2023-11-22:
I am experiencing same issue, nvidia-container-toolkit is also installed.

hemalshahNV, 2023-11-30:
We're looking into this but haven't been able to reproduce this yet with the same OS and Docker version. We're still running a few more experiments on freshly provisioned machines to see if we can narrow it down.

Our theory is that setup instructions in nvidia-container-toolkit is different than what our machine provisioning scripts do (listed below):

```
# Install Nvidia Docker runtime
curl -s -L https://nvidia.github.io/nvidia-container-runtime/gpgkey | \
  sudo apt-key add -
distribution=$(. /etc/os-release;echo $ID$VERSION_ID)
curl -s -L https://nvidia.github.io/nvidia-container-runtime/$distribution/nvidia-container-runtime.list | \
  sudo tee /etc/apt/sources.list.d/nvidia-container-runtime.list
sudo apt-get update
sudo apt-get install -y nvidia-container-runtime
sudo systemctl restart docker

sudo gpasswd -a $USER docker
sudo usermod -a -G docker $(whoami)
newgrp docker
```

mrlreable, 2024-05-24:
Hi,

is there any update regarding this issue? I'm experiencing the same on Ubuntu 22.04, Docker v4.30.0

sid-isq, 2024-05-28:
facing the same issue

EmanuelCastanho, 2024-05-28:
The previous solution did not solve my problem. 
My original `daemon.json` was:
```
{
    "runtimes": {
        "nvidia": {
            "args": [],
            "path": "nvidia-container-runtime"
        }
    }
}
```

I changed for the above one and did not solve. I already installed nvidia-container-toolkit. I am using Ubuntu 22.04.3 LTS.

tanelikor, 2024-05-28:
Happened to run across this thread, so will give my experience: 

I had the same problem a couple weeks ago, also with Ubuntu 22.04. I had docker installed via snap, and that caused some of the paths to be different than what the Nvidia tools expect. I'm sure it should be fixable for the snap installation as well, but for me the easiest solution was to remove docker entirely and re-install it via apt-get as instructed [here in docker guides](https://docs.docker.com/engine/install/ubuntu/). I tried to make it work with the snap version but quickly ran out of patience and decided to just reinstall docker entirely.

So if you haven't already, you might want to check how your docker is installed.

evstigneevnm, 2024-07-26:
I faced the same problem with the installed NVIDIA container toolkit. It helped me to reconfigure the runtime
`sudo nvidia-ctk runtime configure --runtime=docker`
and then restart the daemon:
`sudo systemctl restart docker`

It worked for me. Manual editing of `/etc/docker/daemon.json` and then restarting the daemon didn't, for some reason.

hipop, 2024-08-08:
I encountered the same problem and successfully solved it.
My environment is "Win10 + wsl + Docker Desktop".
The daemon.json file is not only exists in the "/etc/docker/daemon. json" but also in the Windows host directory : "C:\Users\XXX\.docker\daemon.json". 

So, Editing the file C:\Users\XXX\.docker\daemon.json to include:
```
{
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    }
}
```
and then running in wsl :
```
sudo systemctl daemon-reload
sudo systemctl restart docker
sudo docker run --rm --runtime=nvidia --gpus all ubuntu nvidia-smi
```

I hope this can help you.

yfyang86, 2024-08-12:
I get the corner case repeated several times:

env:

- Ubuntu 24.04
- Docker: Docker version 27.0.2, build 912c1dd
- CUDA: 12.x
- Toolkit: 1_1.13.5-1
- Installation: [install-guide](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html) 

## Bad case 

If one enables non root user to use docker like this:

```bash
sudo groupadd docker
sudo gpasswd -a $USER docker
```

and manually config the  `/etc/docker/daemon.json` like this:

```json
{
    "default-runtime": "nvidia",
    "runtimes": {
        "nvidia": {
            "args": [],
            "path": "nvidia-container-runtime"
        }
    }
}
```

1. After `sudo systemctl restart docker`. One may 

- Get the right feedback with `sudo docker run --rm --runtime=nvidia --gpus all ubuntu nvidia-smi`;
- Get `docker: Error response from daemon: unknown or invalid runtime name: nvidia.` error message   with `sudo docker run --rm --runtime=nvidia --gpus all ubuntu nvidia-smi`.

2. If one follows `Rootless mode` session

- One should set up `$HOME/.config/docker/daemon.json` manually (well, copy `/etc/docker/daemon.json` might work);
- Get the following error message when execute `sudo nvidia-ctk config --set nvidia-container-cli.no-cgroups --in-place`:

```
No help topic for 'config'
```

I think the above configuration is not covered  in this setting (at least for version `1_1.13.5-1`) and takes guarantee that it is executed with `root`/`sudo` privilege or some kind of `Rootless` mode.

taylor-artemenko, 2024-09-15:
> I encountered the same problem and successfully solved it. My environment is "Win10 + wsl + Docker Desktop". The daemon.json file is not only exists in the `/etc/docker/daemon. json` but also in the [...]

Thank you for this, I am using the WSL integration with Docker Desktop on windows and totally forgot about the config on the windows host, this fixed it for me.

PeisenWong, 2024-09-28:
i also have the same problem. I think there has some path issue for the ~/.docker.
`rm -rf ~/.docker`
After removing the ~/.docker, it works for me, but since I downloaded the docker desktop, the ~/.docker will appear everytime, I do not sure this is the best solution or not

ghaeflig, 2024-10-28:
I installed docker desktop for linux and I did not manage to get nvidia runtime. Then, I changed docker context to the default one and it works. It seems that for linux you have to use docker engine and not docker desktop for gpu support. Docker desktop points to the ~/.docker folder.

`docker context ls`
`docker context use default`

whoslcy, 2025-08-01:
> i also have the same problem. I think there has some path issue for the ~/.docker. `rm -rf ~/.docker` After removing the ~/.docker, it works for me, but since I downloaded the docker desktop, the ~/.docker will appear everytime, I do not sure this is the best solution or not

How about this?
```
nvidia-ctk runtime configure --runtime=docker --config=$HOME/.docker/daemon.json
```