NVIDIA-ISAAC-ROS/isaac_ros_visual_slam #138: Feature Request to add support for RGBD stream

sandeepdutta, 2023-12-15:
Hello,

This is a feature request to add support  RGBD images instead of Stereo images, if it not very difficult.
I am assuming that you convert the stereo images to a disparity map early in the processing pipeline.

Thank you
Sandeep

jaiveersinghNV, 2023-12-19:
Thank you for your feedback!

The stereo image processing step is a core part of the cuVSLAM library, and a simple depth map would be insufficient for this algorithm. We'll keep your request of a monocular RGBD SLAM solution in mind for the future.

zhangzscn, 2023-12-26:
At present, there are few binocular camera models, but there are many RGBD camera models. In addition, some binocular cameras will add speckle emission hardware due to the hardware structure design, resulting in the binocular camera being unavailable. Can you add the SLAM algorithm for RGBD? @jaiveersinghNV